PiSupply/PiJuice #339: Add timer before/after triggering SYS_FUNC functions

Tijuweb, 2019-07-23:
I use the PiJuice HAT for my RPi3 and I can tell it's a must have!

However, I want my RPi3 go to deep sleep but not immediately after a power loss (in case of short power lost). I want to have a timer and determine at the end of this timer if there is still no power.
Knowing this, I can use a function and attached it to the no power interrupt provided by the PiJuice.There is my script
scrpit1.py : 
```
#!/usr/bin/python3

from pijuice import PiJuice # Import pijuice module
from subprocess import call # Import subprocess

pijuice = PiJuice(1, 0x14) # Instantiate PiJuice interface object
print(pijuice.status.GetStatus()) # Read PiJuice status
print('No more power -')
# Start a thread with a timer 

# THREAD_TIMER
# if there is still no power at the end of the timer, go to deepSleep mode as SYS_FUNC_HALT_POW_OFF interrupt would do
``` 

First, the Rpi doesn't go into real deepSleep since the PWR led is still ON. 
Secondly, the Rpi doesn't wakeup when power is back since it's not a SYS_FUNC_HALT_POW_OFF interrupt. I need to use the SW1 button and the 

I've enable "No power" : USER_FUNC1 trigger and "wakeup_on_charge" trigger also.

```
{
  "user_functions": {
    "USER_FUNC15": "",
    "USER_FUNC4": "",
    "USER_FUNC13": "",
    "USER_FUNC12": "",
    "USER_FUNC5": "",
    "USER_FUNC10": "",
    "USER_FUNC7": "",
    "USER_FUNC9": "",
    "USER_FUNC1": "/home/pi/pijuice/script1.py",
    "USER_FUNC3": "",
    "USER_FUNC11": "",
    "USER_FUNC6": "",
    "USER_FUNC14": "",
    "USER_FUNC8": "",
    "USER_FUNC2": ""
  },
  "system_events": {
    "low_charge": {
      "enabled": false,
      "function": "NO_FUNC"
    },
    "no_power": {
      "enabled": true,
      "function":  "USER_FUNC1"
    },
    "button_power_off": {
      "enabled": false,
      "function": "NO_FUNC"
    },
    "watchdog_reset": {
      "enabled": false,
      "function": "NO_FUNC"
    },
    "forced_power_off": {
      "enabled": false,
      "function": "NO_FUNC"
    },
    "low_battery_voltage": {
      "enabled": true,
      "function": "SYS_FUNC_HALT_POW_OFF"
    },
    "forced_sys_power_off": {
      "enabled": false,
      "function": "NO_FUNC"
    }
  },
  "system_task": {
    "watchdog": {
      "period": 4,
      "enabled": false
    },
    "min_charge": {
      "enabled": true,
      "threshold": "7"
    },
    "enabled": true,
    "ext_halt_power_off": {
      "period": "30",
      "enabled": false
    },
    "wakeup_on_charge": {
      "enabled": true,
      "trigger_level": "5"
    },
    "min_bat_voltage": {
      "enabled": false,
      "threshold": 3.3
    }
  }
```

Is there a possibility to trigger the same functions like SYS_FUNC_HALT_POW_OFF in another thread? 

Regards.

tvoverbeek, 2019-07-23:
Just enable "ext_halt_power_off" in system task.
At the end of your timer execute `subprocess.call(["sudo",  "halt"])`
This should shutdown the Pi and after 30 seconds (can be reduced to 20) remove the 5V power to the Pi and put the PiJuice in low power mode.

Tijuweb, 2019-07-25:
Dear @tvoverbeek,
Thank you for sharing this solution. The problem is when the power is back on USB, the PiJuice doesn't start the Rpi, I need to wake it up with the SW1 button. 
I would like to know if it's possible to trigger the SYS_FUNC_HAL or SYS_FUNC_HALT_PWR_OFF in my own script to be sure the Rpi wakes up when power gets back (using the 'wake up on charge' feature). As you mentioned here : 
https://github.com/PiSupply/PiJuice/issues/299#issuecomment-476291108_

> If you execute the shutdown yourself, the 'wakeup on charge' setting is not copied to the PiJuice.

Do you plan to add this feature to other situations in order to integrate it in our own script?

tvoverbeek, 2019-07-25:
You are right the 'Wakeup on charge' is not set in the PiJuice when not using SYS_FUNC_HALT or SYS_FUNC_HALT_PWR_OFF.
What you can do in your own script after the timeout is to set it yourself before the `subprocess.call(["sudo", "halt"]`:
`pj.power.SetWakeUpOnCharge(0.0)` (pj is the result from pj=pijuice.PiJuice(1,0x14))
0.0 is the chargelevel to trigger the wakeup. Change it if you need a different level.

Tijuweb, 2019-07-26:
It works as expected! I set the WakeUpOnCharge to 0.0 and now I can wake up my Rpi when power is back after using the Halt command! 
Thank you!
Regards

closed 2019-07-26 by Tijuweb