PiSupply/PiJuice #343: Over Temperature Recovery

AlanHares, 2019-07-30:
Hi I was wondering what happens when / if the Raspberry Pi should halt due to over heating of the CPU, will the pi juice try to restart the Pi because of the watchdog or does the pi juice know that the Raspberry Pi halted due to cpu over temperature?

Is there a way we can monitor the cpu temp while the Pi Juice is in low power / sleep mode ?
this way we could restart the Raspberry Pi when the cpu temp had lowered.

AlinNZ

AlanHares, 2019-08-01:
anyone there ??

tvoverbeek, 2019-08-02:
The Pi will first reduce clock speed (throttling) when the cpu temperature rises. I have never heard (yet) of the Pi halting. The CPU can easily run at 80 deg.C
Anyway if the Pi halts for any reason the PiJuice watchdog will try to do a reset (provided it is enabled). The watchdog is reset every time an I2C communication occurs.
Note at high battery temperatures (> 45-50 deg) the battery will not charge any more.
In the case of a PiJuice HAT with BP7X mounted on top of the Pi you will have problems with the PiJuice long before the Pi has over temperature problems.

You would need external CPU temperature measurement which gives a signal when the CPU temperature is low enough. Then you can use the new wakeup trigger via the IO2 pin on the PiJuice.
From https://github.com/PiSupply/PiJuice/blob/master/Software/Source/debian-base/changelog for 1.5:
```
    - Use IO2 digital input as source for wakeup enabling peripheral attached to
      the P3 header to wake-up Raspberry Pi
```

AlanHares, 2019-08-03:
Thanks @tvoverbeek thats great, I need to be able to start / shutdown the pi based on the state of the micro USB input, when power is "Present" on the micro USB I need the Pi to start up, when Power is "Not Present" on the micro USB I want the Pi to Shutdown and power off.

Is this possible with a script of some sort and if so could you give me an example of what the script might look like please?

Kind Regards AlinNZ

tvoverbeek, 2019-08-04:
To shut down the Pi when no external power is present enable the 'No power' on the System events tab and connect it to the appropriate function, e.g. SYS_FUNC_HALT_POW_OFF (which shuts down the Pi and removes the 5V power after 30 sec, provided the system task is running) or your own user function (in that case use the 'Software Halt Power Off' option on the System Task pane which also removes the 5V power after a settable delay with a minimum of 20 seconds)

To boot the Pi when external power returns use the 'Wake up on charge' option on the System Task pane with a trigger level of 0%.

AlanHares, 2019-08-08:
Thanks yes we tried that and we used the code you put on #160 the problem is that if the power is still present at the micro USB when the PJ runs the script it boots the RPI back up immediately due the wake up on Charge setting we have got around this by changing the way we operate and shutdown the system, it would be good if there was an option that allowed for the controlled shutdown from the script but ignored the wake up on charge until the power had been removed and then reapplied (hope that makes sense) this would stop the immediate reboot.

Thanks for all your help always appreciated :)

tvoverbeek, 2019-08-18:
@AlanHares You could use a user script to run on 'No Power'. See my comment here: https://github.com/PiSupply/PiJuice/issues/357#issuecomment-522326448
Modify the script as needed.