PiSupply/PiJuice #345: wakeup/sleep PiJuice from accelerometer

mirakamali, 2019-07-31:
I have a LIS3DH accelerometer and I would like to use it the accelerometer position for making my pi in sleep mode or wake mode. My pi is connected to the pijuice. I understood that the pijuice can wake up from charging or rtc. is their a way to control the pijuice to put my pi on sleep mode(NOT COMPLETE SHUTDOWN) and wake mode (to re-run the script) via an accelerometer position(if acceleration.z is between 0.8 and 1 gor eg.)
thanks in advance

shawaj, 2019-07-31:
You would need to connect it to the pins on P3 IO1 and 2.

https://github.com/PiSupply/PiJuice/blob/master/Hardware/README.md

These do have some software functions over i2c:

https://github.com/PiSupply/PiJuice/blob/master/Software/README.md

But for what you are wanting to do you would need to use customised firmware.

mirakamali, 2019-08-02:
thank you for your answer, however what do you mean by customized firmware? is their a sleep function that makes the pi on sleep mode? is it ok if you explain to me a bit more. thank you.

tvoverbeek, 2019-08-02:
Forget about writing a customised firmware. The PiJuice firmware is the code running in its micro controller (MCU). It is a complex beast. If you want the source for the current 1.3 firmware is here; https://github.com/PiSupply/PiJuice/blob/master/Firmware/Sources-V1.3_2019_02_28.zip

You probably should be able to do something with the available interrupts from the LIS3DH.
The Pi does not have a sleep mode. The only thing you can do is minimize activity by e.g. waiting for an interrupt to start your script instead of polling the accelerometer.
Have a look af the Adafruit tutorial for the LIS3DH breakout https://learn.adafruit.com/adafruit-lis3dh-triple-axis-accelerometer-breakout and the LIS3DH datasheet for some ideas.

mirakamali, 2019-08-05:
ok thank you for your answer

closed 2019-08-06 by mirakamali