PiSupply/PiJuice #347: code to kill power to usb when pijuice shuts down

giff1000, 2019-08-02:


giff1000, 2019-08-02:
Hi everyone, sorry this is a quick question; could someone tell me the code to use when shutting down the pijuice in order to stop power supply to usb - i have a 3g project and just realised that the dongle still has power even when the pijuice has shut the pi down.

Many thanks 

Kind regards

Paul

ChristopherRush, 2019-08-02:
It sounds like you are powering the PiJuice using the Raspberry Pi micro
USB rather than directly on the PiJuice. If your primary power is connected
to the PiJuice, when you do a hard shutdown it will remove power to the
GPIO pins this removing power completely to the Raspberry Pi. How are you
currently shutting down the Pi?

markb9919, 2019-08-02:
Hi

I have been using this command to shutdown the power to the dongle just before the scheduled shutdown.

echo 0 | sudo tee /sys/devices/platform/soc/20980000.usb/buspower >/dev/null

When the device starts up it restores the power to the usb.

tvoverbeek, 2019-08-04:
I assume you schedule the shutdown using a cron job or similar.
In that case enable the 'Software Halt Power Off' option on the System Task pane. After the delay (minimum 20 seconds) the 5V from the PiJuice to the Pi will be removed and hence the usb bus will have no power. To be safe you can leave your own usb power shutdown active.

closed 2019-10-23 by ChristopherRush