PiSupply/PiJuice #348: pijuice + poe

bluedark, 2019-08-05:
Hello,
I followed this tutorial to feed the battery pijuice in poe with the module hat to add on the pijuice. It works well as soon as I turn on the system, the battery is well supported and there is a network communication. But as soon as I unplug the network cable or there is a micro power cut, the poe module can no longer recharge the battery. FYI, the single poe module on the raspberry works very well. I tried with several raspberry.
Thank you in advance.

shawaj, 2019-08-05:
Do you mean if you unplug it and then plug it back in again?

Have you got the jumper on the PoE switch HAT on or off?

bluedark, 2019-08-05:
Yes.
I never put a jumper on my poe module. 
It seems to me that it is used to physically boot the raspberry with the button, right?

shawaj, 2019-08-05:
What happens if you perform the same test, but without the PiJuice attached?

bluedark, 2019-08-05:
As I said, without the pijuice, the poe module works fine

shawaj, 2019-08-05:
What PoE switch or injector are you using?

bluedark, 2019-08-05:
sorry I forgot to put the link in my first post, I edit it. I use the same one as the one in the link. The orange module for raspberry 2.

shawaj, 2019-08-05:
Yes, you are using this PoE HAT right? https://uk.pi-supply.com/products/pi-poe-switch-hat-power-over-ethernet-for-raspberry-pi

But what I mean, is what PoE injector / midspan / switch are you using on the other side of the ethernet cable?

Thanks

bluedark, 2019-08-05:
oh désolé, j'ai un switch linksys lgs116p.

bluedark, 2019-08-06:
an idea?

bluedark, 2019-08-08:
for information, when I plug other poe equipment in the same place on the same plug with the same cable, no problem ... I tried with a raspberry 2, 3 and 3B +, same result ... Thank you in advance

bluedark, 2019-08-19:
anyone?

bluedark, 2019-08-31:
UP
thanks!

closed 2019-10-23 by ChristopherRush