PiSupply/PiJuice #349: Simple Shutdown and Power Off on low battery not working

MediaMeasure, 2019-08-07:
Good Day,

I am struggling with a simple usecase, 

I have the micro usb power plugged into the Pi 3 usb. My goal is to have the PI ON unless while NO 5V is present and the battery voltage drops below 10% it must shutdown. 

If the battery is at 5% and the system is off. then when 5V is supplied again to the Pi3 it must boot up and stay on even though its less that the 10% threshold. 

Im having multiple weird issues,
1. The pijuice is powering off the PI even when level threshold is not yet reached.
2. When power is back i.e. 5V supply pi starts to boot up then immmediatly pi supply shuts it down!

How can I enable and check VERBOSE logs of why the pijuice is doing this.

This is a new PiJuice bought about a week ago with updated firmware.

tvoverbeek, 2019-08-07:
@MediaMeasure Need some more info. What are your settings on the System Task and the System Events tabs (assuming you are using the GUI) and any user scripts you might be using?
WIth that info I can probably tell you how to proceed.

MediaMeasure, 2019-08-07:
Thanks for the prompt response, I have only installed the cli utility. Currently for testing I have the following settings enabled:
System Task Enabled = checked
Wake up on Charge = 15%
Min charge = 90% (for testing purposes as I cannot wait for the whole battery to drain will set to 10% later)

System Events:
Only "Low Charge" enabled "SYS_FUNC_HALT_POWER_OFF"

Besides this nothing else was altered. 

Basically I want my pi on when their is power irrespective of battery level. and only switch off when their is no power and battery is less than 10%

Thanks.

MediaMeasure, 2019-08-07:
Just to sum up my usecase. 
My biggest concern is that I always want the PI to be ON if there is adequate power. i.e. if there is 5V supply or if their is more that 10% battery.

ChristopherRush, 2019-10-23:
Why not set min charge to 10% and then wakeup charge to 1%. Pi will shutdown when battery is less than 10% and then wakeup when 5V power has been restored.

rdragonrydr, 2019-11-18:
If I may help clarify, are you saying that you wish it to turn _off_ when the power is less than 10%, yet turn on when power is supplied and the battery is above 5%?

And it's turning on and then instantly off again?

This sounds like the system isn't checking for input power. So, when the battery is above 5% but below 10%, the system launches a wake event, and then a shutdown since it's also below 10%. You're looking for some sort of latching or hysteresis behavior, and the hat sounds like it's using a strict cutoff that's not used to the on trigger being _less than_ the off one.

I can't guess what the other issue (#1) is, but this may help you or one of the devs solve this.