PiSupply/PiJuice #350: pijuice_cli::Status fails

ghenzinger, 2019-08-07:
Dear everyone, 

I have the current Raspi4 with 2GB RAM and I installed the pijuice HAT with pijuice_base.
I can open pijuice_cli, but when opening "Status" i geth this errormessage:

`
Traceback (most recent call last): File "/usr/bin/pijuice_cli.py", line 2007, in <module> loop.run() 
File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 286, in run self._run() 
File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 384, in _run self.event_loop.run() 
File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 788, in run self._loop() 
File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 825, in _loop self._watch_files[fd]() 
File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 404, in <lambda> event_loop, callback, self.get_available_raw_input()) 
File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 502, in parse_input callback(processed, processed_codes) 
File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 411, in _update self.process_input(keys) 
File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 511, in process_input k = self._topmost_widget.keypress(self.screen_size, k) 
File "/usr/lib/python3/dist-packages/urwid/container.py", line 595, in keypress *self.calculate_padding_filler(size, True)), key) 
File "/usr/lib/python3/dist-packages/urwid/container.py", line 1590, in keypress key = self.focus.keypress(tsize, key) 
File "/usr/lib/python3/dist-packages/urwid/container.py", line 2271, in keypress key = w.keypress((mc,) + size[1:], key) 
File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress return self._original_widget.keypress(maxvals, key) 
File "/usr/lib/python3/dist-packages/urwid/listbox.py", line 999, in keypress key = focus_widget.keypress((maxcol,),key) 
File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress return self._original_widget.keypress(maxvals, key) 
File "/usr/lib/python3/dist-packages/urwid/wimp.py", line 540, in keypress self._emit('click') 
File "/usr/lib/python3/dist-packages/urwid/widget.py", line 460, in _emit signals.emit_signal(self, name, self, *args) 
File "/usr/lib/python3/dist-packages/urwid/signals.py", line 265, in emit result |= self._call_callback(callback, user_arg, user_args, args) 
File "/usr/lib/python3/dist-packages/urwid/signals.py", line 295, in _call_callback return bool(callback(*args_to_pass)) 
File "/usr/bin/pijuice_cli.py", line 1907, in item_chosen callback() File "/usr/bin/pijuice_cli.py", line 191, in __init__ self.main() 
File "/usr/bin/pijuice_cli.py", line 251, in main status_args = self.get_status() 
File "/usr/bin/pijuice_cli.py", line 206, in get_status general_info += ", %3d°C, %s" % (temp, status['battery']) KeyError: 'battery'`

I2C is enabled
I have the Raspian installed, veriosn 4.19.64

Any ideas, what the problem might be?
Thank you
Gerald

tvoverbeek, 2019-08-07:
@ghenzinger The PiJuice with the current firmware (1.3) has problems communicating with the Pi4
using I2C (only on the Pi4, older Pis work fine). This is due to some change in the I2C hardware on the Pi4. When you use the older I2C module (bcm-2708) thein it should work on the Pi4.
Pu the following line in /boot/config.txt: `dtoverlay=i2c-bcm2708` and reboot.
Hope it works then.

ghenzinger, 2019-08-07:
It works! Thank you!
Gerald

tvoverbeek, 2019-08-08:
Closing the issue

closed 2019-08-08 by tvoverbeek