PiSupply/PiJuice #354: How long can the PiJuice Run when RPi off

MediaMeasure, 2019-08-13:
Good Day

If I understand correctly the PiJuice RTC is powered with the same battery that powers the RPi system.

If I set the standard PIJuice (1820mAh) to shutdown the Pi @ 10% battery level, how long with the PiJuice be able to run before we loose the RTC time?

Tnx

ChristopherRush, 2019-08-15:
In run mode the PiJuice consumes around 5mA. In deep sleep mode this is less than 1mA. For deep sleep mode to occur the 5V GPIO rail has to be off and mains power removed from the PiJuice HAT.

MediaMeasure, 2019-08-15:
Thanks @ChristopherRush,

My calculations would then be as follows:
10% of a 1820mAh battery means I still have 182mAh i.e. I can run in deep sleep for 182 hours or roughly 7 days? 

This is really terrible for an RTC point of view where the standard DS1307/DS3231 running a coin cell(+-200mAh) have expectancies of 20 years+

Is the RTC functionality built into/provided by the ST Micro-controller? 

Thanks.

ChristopherRush, 2019-08-15:
Yes, exactly that. The RTC functionality is built into the St microcontroller so as long as it has power from the battery it will retain the time/date.

MediaMeasure, 2019-08-15:
I honestly think PiJuice Team should consider documenting the limitations of the RTC. In my professional opinion when I saw PiJuice advertised as having a RTC I expected something semi decent. however after basic testing I encountered 3 issues:

1. Sometimes the RTC is not picked up by the PI, i2cdetect shows 68 and not UU.
2. Time precision is not great. I run only off RTC time and found hours missing after a few days (battery maintained at > 90% level).
3. Time retention duration is severely limited, RTC's are suppose to give years of life not days.

With the above points I would hesitate to classify the Clock provided as a Real Time Clock.

ChristopherRush, 2019-08-15:
1. Sounds like a driver issue, are you using a different EEPROM address?

2. Since the I2C drivers are not loaded early on in the boot process the time does not get copied across from the RTC to the OS. Therefore there is a fake hwclock which ensures the time continues where it was at the previous shutdown early in the boot. If you have internet then the time will sync every 15 mins and update the RTC. Otherwise you need to manually copy the time from the RTC to the OS. Everything is well documented here - https://github.com/PiSupply/PiJuice/tree/master/Software#pijuice-rtc

3. This is an issue you will need to take up with ST, nothing to do with us.

closed 2019-09-16 by ChristopherRush