PiSupply/PiJuice #355: no battery icon in system tray

cainkirisgai, 2019-08-15:
after installing the pi juice software i can open the gui and it pulls up all the info but it never loads the system tray icon.

tvoverbeek, 2019-08-15:
The tray icon app is started when you log in as part of the desktop startup. Did you log out and bavk in after the install?

cainkirisgai, 2019-08-15:
multiple times

tvoverbeek, 2019-08-15:
OK. In a terminal on the desktop run the tray app manually: `/usr/bin/pijuice_tray.py`
You will get a few deprecation warnings, but hopefully you also get an error message which might explain why you do not see the tray icon.

cainkirisgai, 2019-08-15:
gives me
Traceback (most recent call last):
   File "/usr/bin/pijuice_tray.py", line 10, in <module>
      import gi
ImportError: No module named 'gi'

cainkirisgai, 2019-08-16:
nm i got it turns out for some reason python3 gi and gtk were not installed had to install them got the icon up and running now, but another question is there anyway to show this icon over say emulationstation or anything like that?

tvoverbeek, 2019-08-16:
@cainkirisgai Short answer: Very difficult.
Emulationstation uses a completely different graphics system. It is written in C++ and uses SDL as graphics driver. You would have to implement it yourself including the I2C communication to get the battery charge level. Of course this would only work in the ES frontend, not in the individuak emualtores (which probably use their own graphics drivers)

tvoverbeek, 2019-08-17:
@cainkirisgai Closing the issue.

closed 2019-08-17 by tvoverbeek