PiSupply/PiJuice #356: Power Interruption Issue

StudentSA, 2019-08-15:
Good Day!

Consider the following edge use case, RPi with external power supplied to RPi directly. 
Pijuice Hat with Python configuration to switch off on low battery:

```
    print("Powering off")
    pijuice.power.SetWakeUpOnCharge(60)
    pijuice.power.SetPowerOff(30)
    os.system('sudo shutdown -h now')
```

When external 5V supply is removed and battery depletes below the threshold the shutdown and power-off is initiated. 

If during this very instant i.e. 1-29 seconds after the shutdown sequence is initiated you find that the external 5V supply is restored to the pi. I have then observed that the pi will boot up again and the PiJuice will charge as normal. 

However if at any time later the 5V Supply to the Pi is once again removed I have found that PI immediately has the GPIO power removed and is HARD switched off, even though the battery is charged to a level above the threshold. 

Please can you advise on why this behaviour is seen and if it is possible to cancel the HARD power off that the PiJuice seems waiting to initiate?

Best Regards

StudentSA, 2019-08-16:
A much more simple test of this,
With external power applied to the RPi and not the Pijuice. 
Trigger a poweroff via python:
pijuice.power.SetPowerOff(10)

After the delay elapses nothing will happen as the Pi is Powered directly, so the Pi still runs.

On removing External power from the Pi the PiJuice does not supply power anymore.

I guess my question is possibly, is their a command or software mechanism to instruct the PiJuice Hat to reapply power i.e. PowerON type command?

Thanks.

StudentSA, 2019-08-20:
@mmilann, please can you advise if the SetPowerOff can be undone? i.e. can I achieve a SetPowerOn that reapplies GPIO power and UPS functionality? 
I have tried  SetPowerOff(255) as 255 seems to be the disabled value. but since the event has already elapsed. I need a way to reconnect the GPIO power drive.

Abraham1220, 2019-10-30:
Could you already solve this issue? I detected the same behavior and also couldn't find in the firmware code an easy possibility to execute "Turn5vBoost(1);" after restart of the PI.

Thanks

StudentSA, 2019-10-31:
Hey @Abraham1220 , No I have not found a solution to this.

Abraham1220, 2019-11-10:
How can we get feedback from supplier vendor team to solve this bug?

@tvoverbeek , is someone reading this from PiSupply?

ChristopherRush, 2019-11-10:
Yes we are. But we cannot deal with every request all at once. We are only
a small team so please be patient with us.

ChristopherRush, 2019-11-14:
I've been unable to solve this. I would have thought enabling WakeUp on Charge and disabling/enabling GPIO input might have solved it. @mmilann Will know if anything can be done in the Firmware otherwise I think a hard reset is needed after initiating `pijuice.power.SetPowerOff(10)`.

Abraham1220, 2020-06-22:
@ChristopherRush, are you aware already of a solution or workaround to enable power again? Still suffering to have no bugfix available.

tnx/a

tvoverbeek, 2020-06-22:
@Abraham1220 I do not know if you are aware of a change present in the 1.4 firmware:
In the case of the first post (power returns during the countdown to turn the 5V from the PiJuice off) now when external power returns during the countdown to turn the 5V from PiJuice off, the countdown is cancelled and the 5V will not be turned off.
You need to rely on wakeup-on-charge for the reboot (as in the original post).
This change was implemented in response to this issue.

From the pijuice-base 1.6 package changelog:
```
  * Firmware V1.4_2020_01_17
    - Reworked I2C code, so using i2c-bcm2708 is not neeed any more for Pi 4.
    - Fixed fluctuating temperature readings.
    - Changed order of battery profiles. PjZero 1000mAh is now profile 0
      (= default on PiJuice-Zero)
    - Delayed power off will be cancelled if RPi 5V power is restored before   <===
      the timer runs out (Issue #356).                                         <===
    - Fixed missing clearing wake-up flags for the GPIO3 wakeup case.
```

Abraham1220, 2020-06-22:
@tvoverbeek, I am already using and testing the new version. There is no improvement in regard to my situation. 

Power is attached to PI's USB and in case external power is lost, I would like to shutdown the PI when the battery charge level is below a certain value (e.g. 15%). Additionally I would like to poweroff with some small delay (e.g. 60 seconds), set a wakeupChargeLevel (e.g. 18%) and shutdown the PI.

Everything is working well in above described scenario. After external power reappears the PI is powered up again, battery is charging and later waking up the PI. **BUT, in case the external power is lost again, the battery is not supplying voltage to the PI anymore.** This is easily replicable.

The only chance to get rid is to unplug USB power and LiPo from PiJuice and reapply in the order: LiPo and afterwards USB power. In case you do it the other way round (USB power and then LiPo) the issue remains.

I tried to understand the firmware source code and find a solution within the existing firmware, but I couldn't do so.

tnx/a

tvoverbeek, 2020-06-22:
@Abraham1220 Can you post the settings and user scripts you use to achieve your scenario?
Then I can try to reproduce what you are seeing.

You write: "After external power reappears the PI is powered up again, battery is charging and later waking up the PI"
If you apply external power to the Pi again, it should start immediately and not by wake-up.

Abraham1220, 2020-06-23:
@tvoverbeek I think it's difficult to post the complete script as I am programming in perl and have written a very complex tool to handle power outage, sync databases, trigger audio alarms and fill logs which are implemented with additional libraries. Therefore I just highlight the main pijuice commands I perform. I hope this is sufficient enough to understand what I am doing,

When I shutdown the PI at BatteryChargeLevel 14% and ask the PiJuice to start the PI again when BatteryChargeLevel reaches 17% then it's doing it properly. So when main USB power comes back the PI is **NOT** immediately starte; only when the battery charge level is >= 17%.

Here are the settings:
[image]
[image]

Here the fragment of the routine:

```
use constant WATCHDOG_TIMEOUT                => 3;
use constant BATTERY_LEVEL_SHUTDOWN          => 14;
use constant BATTERY_CHARGE_LEVEL_WAKEUP     => 17;
use constant POWEROFF_DELAY_SECS             => 60; 

setWatchdog(WATCHDOG_TIMEOUT);
setWakeUpOnChargeLevel(BATTERY_CHARGE_LEVEL_WAKEUP);

while ($gRunning == xTRUE) {

  my @status = readPiJuiceData(PIJUICE_STATUS_CMD, 1, 2);

   if (defined $batteryChargeLevel && $batteryChargeLevel <= BATTERY_LEVEL_SHUTDOWN) {

     setWatchdog(0);
     setPowerOffDelay(POWEROFF_DELAY_SECS);
     setWakeUpOnChargeLevel(BATTERY_CHARGE_LEVEL_WAKEUP);
     callSys('poweroff');

   }

}
```

Does this help to understand the issue?

tnx/a

tvoverbeek, 2020-06-23:
@Abraham1220 Thanks. I'll try to implement this on my reference system (with Python).
I guess you implemented the whole pijuice.py module in Perl. Impressive!!
Question: I suppose you have disabled the pijuice service since the service could interfere with your Perl program
(Your Perl shutdown could also be done with the service).
Correct?

Ton

Abraham1220, 2020-06-23:
@tvoverbeek Correct, the pijuice service is not even installed on my PI. I just use the CLI for testing and reference for coding it in perl. But I didn't convert everything to perl, just the routines I require. So only my perl script is interacting with the PiJuice and is running as a systemd service.

So to avoid the issue I described, I set currently POWEROFF_DELAY_SECS to 0xFF in my script.

tnx/a

StudentSA, 2020-06-24:
@Abraham1220, You mentioned you are supplying the external 5V directly to the PI? i.e. you are not supplying power to the PiJuice Hat?

> Power is attached to PI's USB and in case external power is lost, I would like to shutdown the PI when the battery charge level is below a certain value (e.g. 15%). Additionally I would like to poweroff with some small delay (e.g. 60 seconds), set a wakeupChargeLevel (e.g. 18%) and shutdown the PI.

> Everything is working well in above described scenario. After external power reappears the PI is powered up again, battery is charging and later waking up the PI. BUT, in case the external power is lost again, the battery is not supplying voltage to the PI anymore. This is easily replicable.

Please can you clarify, once external 5V supply is restored you say the PI is powered up? but later waking up the PI?

I think you are actually supplying your external 5V to the Pi Juice Hat and not to the Pi?

Abraham1220, 2020-06-24:
@tvoverbeek I try to explain. I am powering directly the PI 4 using the USB-C port. The pijuice is not directly powered.

Scenario:

1. PI is powered via USB-C, LiPo is fully charged
2. USB-C power is lost
3. PI is powered by pijuice LiPo
4. Battery capacity is low after some time -> my script is shutting down the PI and doing some pijuice settings as described above
5. USB-C power comes back
6. PI is not immediately starting
7. LiPo is charged
8. After battery capacity reaches wakupOnCharge level, the PI is booting (with my settings this takes about 5 minutes)
9. Everything seems fine -> but in case I loose USB-C power again, the PI is not powered by the pijuice LiPo anymore and is "killed" -> and here is again my nightmare

So does it look to me. Maybe I misinterpret something wrong, but I cannot imagine what.

I tried already to switch to PiJuice port powering but found two strange behaviours and therefore switched back to USB-C powering via PI:
- PI restarts without any reason sometimes during operation -> and this is the nightmare I want to avoid
- Battery is constantly charging at 99% (ok, this I resolved with enabling/disabling charging within my script)

Does this make sense to you?

kr/m

tvoverbeek, 2020-06-30:
@Abraham1220 Something is not working correctly. In step 6 the Pi should start as soon as the power is back.
I ran a test on my Pi4/PiJuice combination with the PiJuice service disabled and masked.
Ran the following Python script:
```
from pijuice import PiJuice
import os
import time

pj=PiJuice(1,0x14)

count = 0
while True:
   print(count % 60)
   ret = pj.status.GetChargeLevel()
   if ret['error'] == 'NO_ERROR':
      chglvl = ret['data']
      print(chglvl)
      if chglvl < 85:
         print('shutting down')
         pj.power.SetPowerOff(60)
         pj.power.SetWakeUpOnCharge(90)
         os.system('sudo poweroff')
   count = count + 1
   time.sleep(2)
```
This should be very similar to your Perl code above.

Test 1:
Starting running on battery at a chargelevel of 92%.
The Pi shut down when the chargelevel fell below 85%. The red power led on the Pi went off after the 60 seconds
delay.
Then reapplying power to the Pi USB-C the Pi started immediately.

Test 2:
Again started running on battery at a chargelevel > 85%
Pi shut down at chargelevel 84 as expected.
This time reapplied Pi USB-C power after the shutdown completed (the 10 flashes of the green led) but before the
60 seconds delay expired.
As expected the Pi did not start immediately (because the 5V from the PiJuice was not switched off)
When chargelevel 90% was reached it booted.
Removing the power later the PiJuice continued supplying 5V to the Pi until the chargelevel dropped again below 85%.

Note: Test 2 behavior relies on PiJuice firmware 1.4 See above: https://github.com/PiSupply/PiJuice/issues/356#issuecomment-647448030
On previous firmwares the  final power removal wpuld cause the Pi to do a hard shutdown, since the PiJuice 5V would be off.

I am wondering if your Perl implementation somehow puts the firmware in a state that it does not respond to I2C commands any more. The pijuice.py module does I2C transfers in a separate thread.
Suggest you first try to find out why the Pi does not start in step 6.

Abraham1220, 2020-07-11:
@tvoverbeek I am now a little bit confused, as I retried your python script and had no issue after shutdown and restart. Same behavior as you described.

Afterwards I checked again my perl code, made some minor changes (which I think didn't change anything important) and retried with shutting down at 14% and waking up at 17%. And also here everything worked similar to the python script. Immediately after 5V returns, the PI is starting. When I afterwards loose 5V again, the LiPo battery supports as requested the PI.

So - from current perspective - my initial problem is gone; although I don't understand why.

kr/m

tvoverbeek, 2020-07-11:
Good to hear it is working now.
I am not so well versed in Perl, otherwise I could take a look at your Perl code before and after your last changes.

Abraham1220, 2020-07-11:
@tvoverbeek Other questions to this topic. Would it be now possible to wait until the charge level of the battery reaches a certain value and only then the PI is starting after the shutdown of my script? Or is this impossible when powering the PI via 5V directly?

kr/m

tvoverbeek, 2020-07-11:
@Abraham1220 If you want to do that then the PiJuice has to continue supplying 5V to the Pi. This of course consumes more power from the battery than when the PiJuice 5V supply to the Pi is off, although when the Pi is off its consumption is very very low (only the GPU remains powered). Then when 5V reappears on the Pi USB  the Pi does not see a power interruption and will not start.
In my python script delete the `pj.power.SetPowerOff(60)` line.