PiSupply/PiJuice #357: Always Wakeup on Charge

StudentSA, 2019-08-16:
Good Day Again!

Apologies for all the messages, first time using the PiJuice and I would like to use it in a product of mine, so really need to get things right. 

Is it possible or can we make it possible to configure wake up on charge as a permanent option in the firmware of the Pi Juice. I.E. if external 5V supply is restored to the PiJuice and the PI is off, it should always always always wake up the pi and keep it running?

Right now my understanding is this flag is cleared and needs to be set every time it is triggered. my concern is around some fault that may occur and the flag is not set causing a potential situation where the pi remains off even though power is available.

Thanks!

tvoverbeek, 2019-08-18:
Here an approach which might work for you.
Wakeup on charge only works when the PiJuice 5V is off when charging is detected.
At wakeup the Wakeup on charge flag is cleared so it has to be set again before shutdown.
Assume the 
- Connect and enable the 'No power' event to a user function, e.g USER_FUNC1, on the System Events tab. 
- On the User Scripts tab specify USER_FUNC1 (e.g. /home/pi/userfunc1.py). The script has to be executable and the scripts owner has to belong to the pijuice group. The default user (pi) already belongs to the pijuice group.
- Here an example of a script which tests of there is still no power after 10 seconds and then sets the wakeup-on-charge with trigger level 0 and shuts down the Pi
```
#!/usr/bin/python3

import sys
import logging
import time
from pijuice import PiJuice
import subprocess

logger = logging.getLogger('user_func1')
hdlr = logging.FileHandler('/home/pi/user_func1.log')
formatter = logging.Formatter('%(asctime)s %(levelname)s %(message)s')
hdlr.setFormatter(formatter)
logger.addHandler(hdlr) 
logger.setLevel(logging.INFO)

pj=PiJuice(1,0x14)
logger.info(str(sys.argv))

time.sleep(10)
logger.info('Timeout expired')

status=pj.status.GetStatus()
if status['error'] == 'NO_ERROR':
   data = status['data']
   if (data['powerInput']=='NOT_PRESENT') and (data['powerInput5vIo']=='NOT_PRESENT'):
      logger.info('Still no power, shutting down')
      pj.power.SetWakeUpOnCharge(0.0)
      subprocess.call(['sudo', 'halt'])
   else:
      logger.info('power is back')
``` 
- Also make sure that on the System Task tab the 'Software Halt Power Off' option is enabled. This ensures that when the 'sudo halt' is executed by userfunc1.py the 5V will be shut down after the delay period.

So when the power to the Pijuice goes off the PiJuice will shut down the Pi and after the delay period the 5V goes off and the PiJuice goes to low power mode. Even if the power reappears just after the shutdown has started the Pi will wake up again after the delay specified in the 'Software Halt Power Off' option has expired

StudentSA, 2019-08-19:
I was hoping as a feature, the pijuice firmware could be configured to always "wake up on charge". The flag based approach can lead to an issue in extreme cases where the user script fails for whatever reason.

Your code shed some light for me and I have used aspects to create a @reboot script that sets wakeuponcharge.

StudentSA, 2019-08-19:
To add on why wake up on charge should be allowed in non volatile memory...
If I have External 5V Supply connected to the PiJuice and the battery runs completely out.

When supply is restored the pi does not wake up on charge. leaving it dead until manual intervention is done (sw1). which is not possible in remote environments.

tvoverbeek, 2019-08-19:
Agreed. In the Firmware source there are commented out parts where the wakeup-on-charge was stored in NV memory. @mmilann Why was this removed?

devicenull, 2019-09-04:
I was fairly confused by this behavior as well, I expected some combination of the flags to let the Pi automatically turn back on when power is restored.

Krenom, 2019-10-30:
Bumping this.

We want to add the PiJuices into our IoT devices that will be out in the field. We currently have concerns that if something goes wrong during startup - e.g. our UPS container doesn't load properly in order to do the setup - that we'll leave our devices in a state that would mean someone physically has to open the case and press the power button just to get the device to turn on!

A complete waste to send someone out to do it, and completely unreasonable to ask, or even want, random shop owners to grab a screwdriver and crack open the devices.

mchollett, 2019-12-06:
I too would like this feature.  Having trouble as described in issue #418.

Deploying these units on a factory floor.  Need to be assured that they will startup when equipment is turned on.  I can't ask operator, maintenance, etc. to go in the panel to press SW1.

Need a reliable way to accomplish this please.

Thanks,
Mark

ThScho, 2019-12-29:
I also need this, please. I expected this to be a standard functionality of a ups.

@tvoverbeek: The suggested solution based on the no_power trigger means that the RPi ist shut down 10 seconds after the 5V from the USB are lost, whilst the mounted battery would be able to keep it alive for hours.

Is that correct?

Regards, Thomas

ThScho, 2019-12-31:
I just performed my own tests – with a surprising result.

### My configuration:
- PiJuice Zero, RPi Zero W
- Battery: LiPo 1000 mAh
- Firmware: 1.3
- System Task:
  - Wakeup on charge: 15%
  - Min charge: 10%
  - Min battery voltage: 3.3
- System Events:
  - Low charge: SYS_FUNC_HALT_POW_OFF
  - Low battery voltage: SYS_FUNC_HALT_POW_OFF

### The test:
1. RPi is running, battery charge level is over 30%, LED blinks blue/green
2. I unplug the power source (5V at PiJuice micro usb), LED becomes green
3. RPi is up and working for hours (depending on battery charge level)
4. Battery gets below the threshold of 10%, LED becomes red
5. After 5 min RPi is shut down, LED shows short red flashes
6. Some hours later I plug the 5V source, LED blinks blue/red
7. After 10 min RPi boots, LED blinks blue/green, charge level is 16%

That is what I'm expecting, I could reproduce this behaviour several times.
I did **not** define a System Task for "no power" and I did **not'** call ``SetWakeUpOnCharge(0.0)``.

tvoverbeek, 2019-12-31:
@ThScho The key is you are using SYS_FUNC_HALT_POW_OFF. This will do the SetWakeUpOnCharge(<threshold>) for you with the threshold you define in the cli/gui.
When you do the shutdown yourself (or a script) then you also need to do the SetWakeUpOnCharge().

ThScho, 2019-12-31:
@tvoverbeek: OK, thanks. But the functionality "wakeup on charge", which is missed by others, seems to work (for me) ...

StudentSA, 2019-12-31:
> @tvoverbeek: OK, thanks. But the functionality "wakeup on charge", which is missed by others, seems to work (for me), **after RPi was running on battery for a long time**. I do not have to initiate a shutdown after "no power" as you suggested here: [#357 (comment)](https://github.com/PiSupply/PiJuice/issues/357#issuecomment-522326448)

@ThScho, Until your battery drains to dead/zero... then when you plug in power... nothing happens. Pi is now in a off state until you physically press sw1.

ThScho, 2019-12-31:
> @ThScho, Until your battery drains to dead/zero... then when you plug in power... nothing happens. Pi is now in a off state until you physically press sw1.

Oops. Happy too early. Didn't test this.

OhSoGood, 2020-07-28:
Same as others. I would like a solution to restart the Pi and Pijuice when battery has gone to 0 and afterwards external power comes back, in all cases (normal shutdown or raspberry crash). That's 100% required for remote use.

tvoverbeek, 2020-07-28:
For the battery completely empty case:
Enable the 'No battery switch on'. This will start the Pi even if the battery is at 0.

Krenom, 2020-07-29:
Good morning,

It appears that while there's a 1.4 firmware, there's nothing in the changelogs.txt that I can see relating to anything other than 1.3. Could someone confirm, in exacting detail, the current state of this with the latest 1.4 firmware?

Looking back at some email conversations back in January, there was mention of persisting the wake on power in nv-memory until there is a respose from the host over I2C.

Has this been implemented in 1.4? If it has, is that the exact workings? Every boot we need to reset the flag?
I also recall that the flag should no longer be cleared if there's power and the shutdown timer is running?

Longer explanation:
We are currently running with 1.3 firmware (about to upgrade to 1.4 but we'd like to know what we're getting for it).

Currently, if we want to initiate a shutdown ourselves, we need to:
- Remove power input
- Set wake on charge
- Start shutdown timer
- Remove WD

The first step is a big one, because if the juice detects power in, it clears the flag, so you can't physically do a shutdown while there's power.

Currently, if power has been removed:
- Wait 5m (might get restored, yay)
- Set wake on charge
- Start shutdown timer
- Remove WD

So long as nothing has gone wrong and these steps, specifically, are iniated and followed in our products, this is working.
If the battery gets unseated or anything during transport, or for whatever reason there's some loss to the juice that causes the volatile memory to be cleared (i.e. the wake on charge flag!), then we have to have the installers crack them open to hit SW1 to get them booted - not fun.

How far towards solving these issues and having the devices boot on every application of power does 1.4 bring us?

Thanks.

Krenom, 2020-07-29:
Well, answered my own question now:

Just updated to 1.4 and the only difference (relevant to this issue) is that the Wake on Charge setting doesn't get reset/used while the power off timer is running.

Still stored in volatile memory, gets wiped on battery removal/drain.

tvoverbeek, 2020-07-29:
For battery removal/drain see my comment above your posts.

Krenom, 2020-07-29:
> No battery turn on: If enabled pijuice will automatically power on 5V rail and trigger wake-up as soon as power appears at USB Micro Input and there is no battery. Disabled by default.

Is it just poorly worded, or does it actually mean it will boot on power input? Because that currently says that it'll just boot when there's no battery. We have batteries, but we want it booting. The above doesn't suggest that.

If it is incorrectly worded and what everyone in this thread desires, then, additionally, how would this be set via commands rather than through the GUI? I don't see any reference to this in the software docs. Then again, I also don't see the ability to get the firmware version but I do know that there's a GetFirmwareVersion() call that can be made... I haven't looked through the Python interface script available in a long while and don't have time at this particular moment.

tvoverbeek, 2020-07-29:
It means it will boot the Pi by turning on the 5V from the PiJuice to the GPIO in case it detects no battery (= battery removed or completely drained). It does not mean it will boot the Pi if there is still life in the battery (=the PiJuice firmware is still running).

Firmware version is available in the GUI: https://github.com/PiSupply/PiJuice/blob/master/Software/README.md#pijuice-hat-config-firmware-menu
and in the CLI: https://github.com/PiSupply/PiJuice/blob/master/Software/README.md#firmware

Krenom, 2020-07-29:
Yep, so this version does not address this issue, fair enough. Are there plans to address this in the (near) future? This issue is nearly a year old and there has been no definitive will/will not implement direction provided in that time.

One of your early replies indicated that this was previously implemented and then removed, but I don't see any conclusion to that...
> Agreed. In the Firmware source there are commented out parts where the wakeup-on-charge was stored in NV memory. @mmilann Why was this removed?

We are running light-weight Alpine docker containers, controlling the juice prgrammatically using the provided [pyjuice.py](https://github.com/PiSupply/PiJuice/blob/master/Software/Source/pijuice.py) Python interface, not using the GUI or the CLI.

As it stands currently, we are finalising plans to remove the juice from all our devices as they are introducing more issues than they are solving for us without this feature.

tvoverbeek, 2020-07-29:
I am only a volunteer helping out Pi-Supply. As far as I know there are no plans to move this setting to non-volatile memory.
If this does not fit your use case, please use an other device.
Closing this issue.

closed 2020-07-29 by tvoverbeek

shawaj, 2020-11-15:
@StudentSA @ThScho @OhSoGood @mchollett @devicenull @Krenom just so you know we are looking into this again as part of some hardware and firmware updates in #580 

I appreciate it might be too late for your application but thought it might be of interest.

If you have any other suggestions or improvements that you'd like to see then feel free to let us know over there.

shawaj, 2020-12-24:
See also #612

dima72, 2021-09-03:
Hello, I have opposite requirements:
I need to disable piJuice _WakeUp on Charging_ programmatically. Ofcource, it works in piJuice_cli GUI, if I disable **System Task->Wakeup on charge** checkbox - it does not start on external power appearance. But if **Wakeup on charge** was enabled programmatically by **pijuice.power.SetWakeUpOnCharge(0)** it will start waking up again and again and I can not disable it programmatically. Setting up **pijuice.power.SetWakeUpOnCharge(110)** does not help, I need manually go to piJuice_cli and disable checkbox there.

please advice.

tvoverbeek, 2021-09-03:
To disable use **pijuice.power.SetWakeUpOnCharge('DISABLED')**
See line 937 in https://github.com/PiSupply/PiJuice/blob/master/Software/Source/pijuice.py

dima72, 2021-09-16:
Hi Ton,
pijuice.power.SetWakeUpOnCharge('DISABLED') by some reason has no effect on settings persistence or
atleast there must be applied some another function to save settings. Please let me know if you see how.

For a moment I've discovered that changing settings in **/var/lib/pijuice/pijuice_config.JSON** before device goes to
sleep, saves settings and Wake up on charge can be changed through this file.

This works, but imho not system-wise efficient.

dima72, 2021-09-17:
**non_volatile=True** is what makes it working.
Great, Thanks.