PiSupply/PiJuice #360: Battery Temperature

StudentSA, 2019-08-19:
Hey,

Seeing some irregular temperature being reported by the PiJuice for the battery. Using a standard PiJuice, see below where i query with a few seconds in between queries.

```
Python 3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pijuice import PiJuice # Import pijuice module
>>> 
>>> pijuice = PiJuice(1, 0x14) 
>>> pijuice.status.GetBatteryTemperature()
{'error': 'NO_ERROR', 'data': 132}
>>> pijuice.status.GetBatteryTemperature()
{'error': 'NO_ERROR', 'data': 31}
>>> pijuice.status.GetBatteryTemperature()
{'error': 'NO_ERROR', 'data': 31}
>>> pijuice.status.GetBatteryTemperature()
{'error': 'NO_ERROR', 'data': 132}
>>> pijuice.status.GetBatteryTemperature()
{'error': 'NO_ERROR', 'data': 31}
>>> pijuice.status.GetBatteryTemperature()
{'error': 'NO_ERROR', 'data': 31}
>>> pijuice.status.GetBatteryTemperature()
{'error': 'NO_ERROR', 'data': 132}
>>> pijuice.status.GetBatteryTemperature()
{'error': 'NO_ERROR', 'data': 31}
>>> pijuice.status.GetBatteryTemperature()
{'error': 'NO_ERROR', 'data': 31}
>>> pijuice.status.GetBatteryTemperature()
{'error': 'NO_ERROR', 'data': 30}
>>> pijuice.status.GetBatteryTemperature()
{'error': 'NO_ERROR', 'data': 30}
>>> pijuice.status.GetBatteryTemperature()
{'error': 'NO_ERROR', 'data': 30}
>>> pijuice.status.GetBatteryTemperature()
{'error': 'NO_ERROR', 'data': 31}
>>> pijuice.status.GetBatteryTemperature()
{'error': 'NO_ERROR', 'data': 31}
>>> pijuice.status.GetBatteryTemperature()
{'error': 'NO_ERROR', 'data': 132}
>>> pijuice.status.GetBatteryTemperature()
{'error': 'NO_ERROR', 'data': 132}
>>> pijuice.status.GetBatteryTemperature()
{'error': 'NO_ERROR', 'data': 31}
>>> pijuice.status.GetBatteryTemperature()
{'error': 'NO_ERROR', 'data': 31}
>>> pijuice.status.GetBatteryTemperature()
{'error': 'NO_ERROR', 'data': 132}
```
Thanks

tvoverbeek, 2019-08-19:
Thanks for the report. Saw the same thing on the GUI the other day: temperature occasionally switching from 31 to 132 degrees. @mmilann Bug in the temperature acquisition from the fuel gauge? For me this was on a Pi4 which gets hotter than previous models, so the MCU temperature easily goes over 45 deg with the HAT on top of the Pi4.

the-louie, 2019-08-31:
Have a brand new Pijuice on an old pi3 and noticing the temperature fluctuating a lot.
```
# python3
Python 3.7.3 (default, Apr  3 2019, 05:39:12)
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pijuice import PiJuice # Import pijuice module
>>> import datetime
>>> import time
>>> pijuice = PiJuice(1, 0x14)
>>> while True: "%s - %s°" % ((datetime.datetime.now()).strftime('%H:%M:%S'), (pijuice.status.GetBatteryTemperature())['data']); time.sleep(1);
...
'10:03:16 - 41°'
'10:03:17 - 132°'
'10:03:18 - 41°'
'10:03:19 - 132°'
'10:03:20 - 132°'
'10:03:21 - 132°'
'10:03:22 - 132°'
'10:03:23 - 41°'
'10:03:24 - 132°'
'10:03:25 - 41°'
'10:03:26 - 132°'
'10:03:27 - 41°'
'10:03:28 - 132°'
'10:03:29 - 132°'
'10:03:30 - 41°'
'10:03:31 - 132°'
'10:03:32 - 41°'
'10:03:33 - 132°'
'10:03:35 - 41°'
'10:03:36 - 132°'
'10:03:37 - 132°'
'10:03:38 - 132°'
```

tvoverbeek, 2019-08-31:
In all 3 cases the outlier temperature is 132.
@shawaj @mmilann Is this a problem with reading the temperature from the fuelgauge IC ???

the-louie, 2019-08-31:
During further testing I saw the temperatures switch between 42 and 132 degrees, 43 and 116, and 44 and 116 degrees.

ArjenR49, 2019-09-02:
I have the same here. F/W 1.3, S/W 1.5. I have only seen 132 degrees as the outlier. 
[image]
[image]
The last field also switches between NORMAL and CHARGING_FROM_IN in a similar fast pace.

tvoverbeek, 2019-09-02:
@ArjenR49 Thanks for the report. The corresponding switching between 'CHARGING_FROM_IN' and "NORMAL' is to be expected. When the temperature gets too high, charging is suspended. Hence the status goes to 'NORMAL'. Anyway it means the MCU firmware believes the outlier temperature is real. @mmilann Have you been able to look into this fluctuating temperature reading issue?

ArjenR49, 2019-09-02:
Hi Ton,
Thanks for the explanation. 
A similar seemingly erratically fluctuating reading is the current value on the GPIO INFO line below the battery info line. Perhaps there is a reason for it - I think I even saw a message about it somewhere - but as it is, the current value is of little use. It goes from close to two A to negative values sometimes. I don’t know how the PiJuice measures it. By measuring voltage drop over a small resistor or by using a Hall sensor. I made my acquaintance with modern & practical Hall sensors only last year after hearing about the Hall effect in 50 yrs. ago when I studied electronics. 
The competing ups product from Olmatic uses a few. They are essential for its proper operation. However, even a stray magnetic field from the motor of a small powered down hard disk would make the Olmatic S.USV behave totally erratically. They don’t shield the Hall sensors at all. My RPi with its S.USV HAT was just sitting on the drive because that was convenient. 
Needless to say probably that I didn’t place this current setup with the PiJuice anywhere near a disk drive or magnet. 
Strong currents on the board near such an IC and in a certain direction may also influence it. 
Hopefully there’s another reason for the fluctuations of the current value, though. 
Groeten uit Bergen NH
Arjen
(On the Road)

zt4, 2019-11-30:
Any updates on the "false" temperature readings of 97/105/116/132 deg C?

Here is a 1-minute sample graph, showing the temperature values while charging. The "real" temperature slowly increases to about 40 deg C during the charge, but I believe that the false high readings are unnecessarily slowing down the charging of the battery.

[image]

tvoverbeek, 2019-12-02:
@zt4 Yes, during the high readings the charging is suspended.
There is a beta firmware which supposedly fixes the false readings.
Could you please try with the attached 1.4 beta firmware and report back if it indeed fixes the false readings?
Thanks in advance.

[PiJuice-V1.4_2019_07_11.zip](https://github.com/PiSupply/PiJuice/files/3912921/PiJuice-V1.4_2019_07_11.zip)

ArjenR49, 2019-12-03:
Ton,
Ik ben voor een aantal maanden in Finland en mijn PiJuice is ingepakt en in Nederland, dus helaas kan ik de nieuwe firmware voorlopig niet proberen. 
Groeten uit Helsinki,
Arjen
(On the Road)

zt4, 2019-12-03:
> Could you please try with the attached 1.4 beta firmware and report back if it indeed fixes the false readings?

Thanks for the beta version. I installed this version yesterday, but still see the same results with the high temperature after two charging cycles.

Should I see any different version info via pijuice_cli? It says version 1.4.

Here is the output from the install:

```
pi@raspberrypi8:~ $ pijuiceboot 14 PiJuice-V1.4_2019_07_11.elf.binary
using address 14
Input file PiJuice-V1.4_2019_07_11.elf.binary
Starting bootloader
Starting bootloader old firmware
Sending get command
bytesRead 13  dataLen 11
ack 79
data: b, 10, 92
data received 11
bootloader version: b
commands: 10, 0, 1, 2, 11, 21, 31, 44, 63, 73
erase page count 38
first page erase succcess 0
erase success 0
page count 303
mem data read 256, 256
...
mem data read 256, 256
Page 0 programmed successfully
EEPROM programming finished successfully
Code executed successfully
```

Then I did sudo shutdown now, pulled the power off, and pulled out the battery for a few seconds.

[image]

tvoverbeek, 2019-12-03:
@zt4 Apologies, I attached the wrong firmware, which does not solve the temperature problem.
Could you try with this one:
See also the changelog.
[Change log.txt](https://github.com/PiSupply/PiJuice/files/3918384/Change.log.txt)
[PiJuice-V1.4_2019_10_25.zip](https://github.com/PiSupply/PiJuice/files/3918390/PiJuice-V1.4_2019_10_25.zip)

zt4, 2019-12-03:
@tvoverbeek I just installed that version, and so-far, via pijuice_cli, the temperature/charging status looks stable. I will monitor over the next charging cycle or two and let you know.

zt4, 2019-12-04:
@tvoverbeek No issues with the temperature after installing the new firmware. Thanks for the help!

Graph below with the upgrade completed at about 21:00 and no false temperatures afterwards.

[image]

tvoverbeek, 2019-12-04:
@zt4 Thanks for confirming this firmware solves the erratic temperature problem.
I'll try to get a new release with this formware included.
Closing the isssue

closed 2019-12-04 by tvoverbeek

StudentSA, 2019-12-19:
Hey @tvoverbeek, 

When will this Firmware/patch be officially released?

tvoverbeek, 2019-12-19:
@StudentSA I am discussing with the 'powers that be' (i am only a volunteer, not employed by PiSupply)
to release it asap, maybe even next week. But no guarantees...

StudentSA, 2019-12-19:
@tvoverbeek, Understood, However I might have found a potential issue where I have a python script that moves the RTC to a different address which is failing now after flashing the v1.4 file above. I will provide more details tomorrow.

StudentSA, 2019-12-20:
@tvoverbeek, Im not sure why but this firmware is not working well for me, heres a small snip of an issue faced:
`
>>> from pijuice import PiJuice
>>> pijuice = PiJuice(1, 0x14)
>>> pijuice.config.GetAddress(2)
{'error': 'NO_ERROR', 'data': '0'}
>>> pijuice.config.SetAddress(2,64)
{'error': 'NO_ERROR'}
>>> pijuice.config.GetAddress(2)
{'error': 'COMMUNICATION_ERROR'}
>>> pijuice.config.GetAddress(2)
{'error': 'COMMUNICATION_ERROR'}
>>> pijuice.config.GetAddress(2)
{'error': 'COMMUNICATION_ERROR'}
`

tvoverbeek, 2019-12-21:
@StudentSA Somehow I got my PiJuice HATs in the same situation with firmware 1.4.
Reported addresses are 0. Although when specifying an address different from 68 for the RTC in the GUI the change seems to work (i2cdetect reports the new address), but the reported address in the GUI remains 0. (The GUI also used pijuice.config.Set/GetAddres).
Reported it to @mmilann who is the author of the firmware.

tvoverbeek, 2019-12-23:
@StudentSA 
@mmilann found the problem and provided an updated V1.4 today.
It solved the i2c addresses problem for me.
Can you try this firmware and check if it also solves the problem for you?
[PiJuice-V1.4_2019_12_23.zip](https://github.com/PiSupply/PiJuice/files/3996223/PiJuice-V1.4_2019_12_23.zip)
Thanks in advance.