PiSupply/PiJuice #362: RTC Issue

ncomettoro, 2019-08-20:
Hi,
i'm working with pijuice setting the wake-up for different period. After the wake-up i enable the flag again.  Randomly the pijuice doesn't wake up even if is powered by the current. I tried with 5 minutes wake up and 1 hour wake up as shown in the pics.
[image]

I tested the wake-up also with every day wake-up but after few months the pijuice doesn't wake up anymore. This test has been made before the update of the firmware. 

Thanks for reply,

tvoverbeek, 2019-08-20:
Is the Pi connected to the internet?
What is the EEPROM address? 0x50?
If so, there is an explanation.

ncomettoro, 2019-08-21:
Thanks fot the reply, the pi sometimes is connected sometimes no. At the begin i connected it to internet so i could sync the date but then i didn't check the connection, theoretically the rtc doesn't need connectio, am i wrong? Where can i find the EEPROM address?

tvoverbeek, 2019-08-21:
You find the EEPROM address in the HAT configuration tab of the GUI. See screenshot below.
[image]
50 is the default address (every official HAT should have an eeprom at i2c address 50).
Raspbian reads the HAT eeprom at 50 (in fact 0x50) at boot.
The PiJuice HAT eeprom specifies to load the ds1339 rtc module.
The DS1339 is the rtc emulated in the PiJuice firmware.
The real DS1339 also has an alarm function.
When you are connected to the internet the timesyncd daemon is running which synchronises the time from the internet to the system clock and to the RTC.
This happens about every 15 minutes. When timesyncd synchronises the time to the (emulated) ds1339 it also **clears** the wake-up rtc-alarm flag.
When this happens while your small program is running after you have set the wake-up flag, the next periodic wake-up will not occur.

Two ways to avoid this:
- Change the EEPROM address to 52. Then the ds1339 rtc module will not be loaded and timesyncd will not clear the wake-up flag. However you lose the synchronisation with internet time.
- Change the EEPROM address to 52 and add the following line to /boot/config.txt: `dtoverlay=i2c-rtc,ds1307`. The ds1307 is identical to the ds1339, but it does not have an alarm function. This way you keep the time synchronisation by timesyncd, but the wake-up flag will not be cleared by timesyncd.

Hope this is a possible explanation for what you experienced

ncomettoro, 2019-08-22:
Hi, 
thanks for the reply, i'm testing this solution and i hope it works. I don't think is an internet synchronisation problem but the last suggest about the wake-up flag could help a lot. 

Thanks, 
i'll Keep you up.

shawaj, 2020-10-24:
@ncomettoro any update? Is it all working now?