PiSupply/PiJuice #363: PiJuice info displayed on a OLED

emosbek, 2019-08-22:
Hello.
Wanted to know if there are any resources or even possible to display stuff like charge level and other battery related info on a I2C OLED module?

shawaj, 2019-08-22:
Yes it is very much possible but we don't have any resources available or
examples

tvoverbeek, 2019-08-22:
As @shawaj wrote, there are no direct resources available.
The easiest is to use the lowest level python functions from the Pijuice module.
They are (partly) documented af the end of the github software README.
Best reference is the source (https://github.com/PiSupply/PiJuice/blob/master/Software/Source/pijuice.py)
If you can drive your OLED display also from python it should not be too difficult.

For something similar see http://frederickvandenbosch.be/?p=2876
This monitors the presence of power input and displays it on an e-paper display.

shawaj, 2020-10-24:
@emosbek did you manage to get this working?

shawaj, 2020-12-05:
No reply in over a month so closing now.

@emosbek feel free to reopen if you don't agree

closed 2020-12-05 by shawaj