PiSupply/PiJuice #364: Adding System functions to headless PiJuice

sensemonster, 2019-08-22:
Where is the best source of information on how to add system functions using just CLI please.  I am trying to add afunction to be run when there is power interruption to the USB power feed.
I know this is straightforward when using the Desktop interface but I can't easily do that in my project as the board is buried in a case.

tvoverbeek, 2019-08-22:
What version of the software are you running?
The pijuice_cli in the current pijuice-base package has the same functionality as the
desktop interface.
On the initial pijuice_cli screen
[image]
select System Events
[image]
and connect a user function to the no power event.
The user function can be specified on the User Scripts page.
[image]

sensemonster, 2019-08-22:
I have the latest version of the software.  I will try your suggestion and thank you.  I was assuming there would be a directory into which I could write the script :-)

sensemonster, 2019-08-22:
Further to my last message I have set things up so that I get a power OFF notification if power is lost.   Is there a hook/event or something similar that I could use to detect the moment the power is restored please?

At the moment my best idea is to use the power OF system event to execute a cron job every n minutes to check when the USB supply status changes and then disable the cron entry.

tvoverbeek, 2019-08-23:
Seems the best solution for relatively short power interruptions (as long as the PiJuice battery can keep the system going).
As a safety net you might want to enable the low charge event and connect it to SYS_FUNC_HALT_POW_OFF as in the earlier screenshot In addition enable the 'wakeup on charge' on the system task screen with a trigger level of 0 (When using pijuice_cli the lowest possible trigger level is 1 due to a bug in the code).
This ensures the RPi will be properly shut down when the battery is almost empty and start up again when the external power returns.

sensemonster, 2019-08-23:
Thanks and yes indeed I already have the low charge and wake up on charge at a certain level.  So that takes care of the things as you have described.   Basically my project is to record how long there has been a power outage.  Therefore, yes, as you have commented, my noob idea depends on battery power keeping the pi running during short power outages.  However the "perfectionist" part of my brain is thinking if there is a GPIO based event that could trigger mu "powerBackON" script :-)

tvoverbeek, 2019-08-23:
Closing thread

closed 2019-08-23 by tvoverbeek