PiSupply/PiJuice #365: External Supply Question

StudentSA, 2019-08-22:
Good Day,

Please can you assist me with a question on supplying the Pi via the GPIO. 

I have a Meanwell 5V SMPS ([Datasheet](https://www.mantech.co.za/Datasheets/Products/SD-15-MW.pdf)) that I use to power the Pi through the GPIO header pins, 4 and 6 i.e. 5V and GND. 

I have noticed that when the Meanwell 5V Supply is off, the Pijuice supplies the Pi and somehow backfeeds into the SMPS illuminating the LED in the SMPS device. 

What would be the correct way to prevent this from happening? I have purchased a 1N5821 Diode that I would like to place between the SMPS and the PI GPIO. would this be okay? My concern is that when Voltage is delivered from the SMPS it will be at Vout-0.5V as the diode has Vf drop. Will the PiJuice still know that Power is available from the GPIO?

Thanks

tvoverbeek, 2019-08-22:
I would connect the 5V SMPS output to the PiJuice input: either the USB or the J4 connector (you will need to add a 2.54 mm 2-pin header).
Then the PiJuice will supply the Pi via GPIO pins 4 and 6.
No need for a diode then.

StudentSA, 2019-08-22:
@tvoverbeek , I agree, however I am using the PiJuice in remote solutions with very limited access and I cant take the chance as per issue #357 where I could be left with a pi that does not boot.

tvoverbeek, 2019-08-22:
OK. According to the SMPS datasheet you can adjust the output voltage to 5.5V. WIth the diode voltage drop you will still have 5V. Should work then.
Just curious, what is your SMPS input voltage? 12V car battery?

StudentSA, 2019-08-23:
@tvoverbeek, My system has a bigger 24V DC supply this SMPS is just DC-DC converting to the Pi's 5V. 

@tvoverbeek, Without the Diode Ive been using the "fine tune knob" on the SMPS to set the voltage to 5.2V (sounds like a safe bet) and charging seems to work okay. 
However after adding the Diode and "Fine Tuning" the SMPS such that Vout is 5.7V with the same 5.2V at the Pi GPIO. I see that the pijuice_cli status indicating that the GPIO power input is "WEAK" and does not charge above 80% now. Any ideas?

tvoverbeek, 2019-08-23:
Cannot tell why you get "WEAK" with the diode. Maybe the output of the DC-DC converter has a larger ripple when at 5.7V.

StudentSA, 2019-09-09:
@tvoverbeek , Just to close up on this issue. The "WEAK" status was due to an issue on my side. I was using the jumper wire you normally buy that looks like ribbon cable to power the Pi through the GPIO. Turns out not all wire is made equally and I needed to change it for a more decent AWG20 gauge wire to carry the current (i.e. 1.5A charging plus the 0.5+ for the Pi).

closed 2019-09-09 by StudentSA

tvoverbeek, 2019-09-09:
@StudentSA Thanks for the feedback on the wire gauge. Might help some other people.