PiSupply/PiJuice #369: Min Charge triggers event even when power is present

sitzmar, 2019-09-01:
While testing, I noticed that the 'Min Charge' system task will trigger an event even when power is supplied. 

**Steps to reproduce:**
1. Enable System Tasks
2. Enable Min Charge, threshold 10%
3. Enable Low Charge System Event, SYS_FUNC_HALT_POW_OFF
3. Partially deplete battery, around 30% for example
4. Plug power back into rpi usb port
5. Modify Min Charge to a value greater than the current charge level
6. Apply setting
7. System shuts down

**Enhancement request:**
Would it be possible to add an option or change the logic so that the min charge event only fires when threshold is exceeded AND there is no power present?

**Use case:**
If you are using the pijuice as a UPS, most would configure it to shutdown the rpi when min threshold is reached, and then 'wakeup on charge' to turn the rpi back on when power has been restored. This works well as long as you are supplying power to the pijuice directly.

If you are supplying power to the rpi usb port, when power is restored the system powers on and then pijuice_sys immediately shuts it back down. Having the option to only trigger the event when no power is present would avoid an unnecessary shutdown.

Other than this one issue, I've been extremely pleased with the pijuice.
Thanks for making such a great product!

StudentSA, 2019-09-02:
Just dropping some honesty here...
I don't know whats up at PiJuice HQ but it does not look like the devs are doing any updates here presently. I have the same requirement and found is easier to just write your own Python code to manage the PiJuice Hat, poll battery level and take action etc.

I have the following script (pijuice_management.py) running on a 5min cron:
```
#!/usr/bin/python3
from pijuice import PiJuice # Import pijuice module
import datetime, time, os, sys

now = datetime.datetime.now()
print (now.strftime("%Y-%m-%d %H:%M:%S"))

#Configure variables
low_batt_threshold = 20 #20
wake_up_on_charge_threshold = 22 #0 means always wake up, +2 from low_batt preferred
power_off_delay_sec = 30 #Delay until power is cut to the PI

#Instantiate variables
power_on=1
battery_level=100

#Ensure the I2C bus has been instantiated
while not os.path.exists('/dev/i2c-1'):
    time.sleep(1)

#Always returns a object to PiJuice, even if module removed.
pijuice = PiJuice(1, 0x14) # Instantiate PiJuice interface object

def get_power_status():
    power_on=1
    status = pijuice.status.GetStatus()
    #print(status)
    if (status["error"] == "NO_ERROR"):
        data = status["data"]
        if (data["powerInput5vIo"] == "PRESENT"):
            power_on=1
        elif (data["powerInput"] == "PRESENT"):
            power_on=2
        else:
            power_on=0
    else:
        power_on=-1
    return power_on

def get_battery_level():
    battery_level=100
    charge_info = pijuice.status.GetChargeLevel()
    if (charge_info["error"] == "NO_ERROR"):
        #print ("Charge Info Valid")
        battery_level = charge_info["data"]
    else:
        battery_level = -1
    return battery_level

def get_battery_temprature():
    battery_temp=-1
    battery_temp_info = pijuice.status.GetBatteryTemperature()
    if (battery_temp_info["error"] == "NO_ERROR"):
        #print ("Battery Temp Info Valid")
        battery_temp = battery_temp_info["data"]
    else:
        battery_temp = -1
    return battery_temp

def get_wakeuponcharge_threshold():
    wakeup=-1
    pj_wakeup=pijuice.power.GetWakeUpOnCharge()
    if (pj_wakeup["error"] == "NO_ERROR"):
        #print ("Wakeup On Charge Threshold Valid")
        wakeup = pj_wakeup["data"]
    else:
        wakeup = -1
    return int(wakeup)

def get_gpio_enabled():
    enabled=-1
    pwr_in_conf=pijuice.config.GetPowerInputsConfig()
    if (pwr_in_conf["error"] == "NO_ERROR"):
        enabled = pwr_in_conf["data"]["gpio_in_enabled"]
    return int(enabled)

def set_GPIO_input(onoff):
    pwr_in_cfg=pijuice.interface.ReadData(0x5E, 1)
    if pwr_in_cfg['error'] == 'NO_ERROR':
        d=pwr_in_cfg['data']
        w = (d[0] | 0x02) if (onoff) else (d[0] & 0xFD)
        pijuice.interface.WriteDataVerify(0x5E, [w])

power_on=get_power_status()
battery_level=get_battery_level()
battery_tempature=get_battery_temprature()
curr_wake_up_on_charge_threshold=get_wakeuponcharge_threshold()
gpio_pwr_enabled=get_gpio_enabled()

#Decision Logic
#Ensure Wakeup on charge Flag is Always set
if (curr_wake_up_on_charge_threshold != wake_up_on_charge_threshold) and (curr_wake_up_on_charge_threshold != -1):
    print("Need to re-enable wake up flag as it is currenly set at :",curr_wake_up_on_charge_threshold)
    pijuice.power.SetWakeUpOnCharge(wake_up_on_charge_threshold)
else:
    print("PiJuice Wake Up On Charge correctly set with value:", curr_wake_up_on_charge_threshold)

#Only if no power is present then we take any action
if (power_on == 0):
    print("External Power supply down. Battery level at:", battery_level,"%. Battery threshold set at:", low_batt_threshold, "%. Temp at:",battery_tempature,"degC")
    if (battery_level <= low_batt_threshold):
        print("Low Battery, need to alert and shutdown")
        #Double Check to ensure their is no power still and level is low then switch off
        if (get_power_status() == 0 and get_battery_level() <= low_batt_threshold):
            print("Powering off")
            pijuice.power.SetWakeUpOnCharge(wake_up_on_charge_threshold)
            pijuice.power.SetPowerOff(power_off_delay_sec)
            os.system('sudo shutdown -h now')
            sys.exit(0)
    else:
        print("Battery Level OKAY at" , str(battery_level))
elif (power_on==1):
    print ("External Power supply up via Raspberry Pi, battery level at:",battery_level ,"% and temp at:",battery_tempature,"degc. No need to take any action")
elif (power_on==2):
    print ("External Power supply up via Pijuice, battery level at:",battery_level ,"% and temp at:",battery_tempature,"degc. No need to take any action")
elif (power_on==-1):
    print ("Unable to communicate with pijuice, running manual mode, no need to take any action")
else:
    print ("Invalid Power State???")

if (power_on==1 and gpio_pwr_enabled==0):
    print("Error state, where power is ON but we are not charging?")
    set_GPIO_input(True)
    print("Charging set to enabled")
```

tvoverbeek, 2019-09-02:
@StudentSA Nice script. Only needs the pijuice module. No need for gui/cli system task (service) and the charge icon.
@sitzmar You can obtain similar results as in @StudentSA's script. The Min Charge triggers the function defined in the System Events tab for Low Charge. If you trigger a user script (for ideas see @StudentSA's script) which tests if the external power is not present and then triggers a shutdown.
Typically you would set the wakeup-on-charge threshold a bit above the low charge threshold.

Note: I am only a volunteer (not employed by PiSupply) interested in helping them out. I can do something on the RPi side of things, but nothing on the firmware.

NickBuilder, 2020-03-18:
I agree, nice script @StudentSA! And I'm also surprised that this kind of functionality wasn't there to start with.

One thought regarding the script execution though. By only relying on 5 minute executions won't there be a slight risk that there will be a "hard" power off if the cron isn't executed before the battery is depleted?

An idea would be to also call the script from the system events "Low charge", "Low battery voltage" and "No power".
Thoughts on this adjustment?

NickBuilder, 2021-07-10:
Could someone please give me some guidance with this feedback?

> [12:19:30] pi@myhabpi:~$ ./pijuice_management.py 

> 2021-07-10 12:19:32

> Traceback (most recent call last):

 

>  File "./pijuice_management.py", line 89, in <module>
> curr_wake_up_on_charge_threshold=get_wakeuponcharge_threshold()

>   File "./pijuice_management.py", line 69, in get_wakeuponcharge_threshold    return int(wakeup)

> ValueError: invalid literal for int() with base 10: 'DISABLED'

Is the code in need of some adjustments or what is going on? I've used the code from @StudentSA.

Thanks!

tvoverbeek, 2021-07-10:
Need more info. Which code is this? Not Python.
Please provide more details, i.e. a reference to the complete code for pijuice_management.py

tvoverbeek, 2021-07-10:
You get this error when wakeup-on-charge is not enabled in the system task.
Modify get_wakeuponcharge_threshold(0 as follows:
```
def get_wakeuponcharge_threshold():
    wakeup=-1
    pj_wakeup=pijuice.power.GetWakeUpOnCharge()
    if (pj_wakeup["error"] == "NO_ERROR"):
        #print ("Wakeup On Charge Threshold Valid")
        if pj_wakeup["data"] == "DISABLED":
            wakeup = -1
        else
            wakeup = pj_wakeup["data"]
    else:
        wakeup = -1
    return int(wakeup)
```

NickBuilder, 2021-07-10:
Thanks! That did the trick.

A small remark on your modification, add a colon to the else statement: "else:"