PiSupply/PiJuice #370: Possible intermittent  RTC issue

sensemonster, 2019-09-03:
HI, I have an otherwise working 10minute interval based project using the PiJuice to wake up and then shut down a full-size Pi.  

Some data is recorded off-site at each 10 minute interval, along with the time.  So, when looking through the data you can see sets of information recorded every 10 minutes as designed.

However, intermittently, approximately every 10-12 hrs the data shows a gap of about 1 hr.

The extract from the data logged is below:

...
2019-09-02T21:09:55.543360500Z    <-- 10mins later as expected
2019-09-02T20:59:04.097690600Z
2019-09-02T20:59:04.073882600Z
2019-09-02T20:59:04.051680600Z
2019-09-02T20:59:04.027161800Z
2019-09-02T20:59:04.004735500Z
2019-09-02T20:59:04.004735500Z   <--  Jump by 1hr (not as expected)
2019-09-02T19:57:15.942077400Z     
2019-09-02T19:57:15.919433600Z
2019-09-02T19:57:15.886807400Z
2019-09-02T19:57:15.851085400Z
2019-09-02T19:57:15.816373600Z
2019-09-02T19:57:15.942077400Z
2019-09-02T19:57:15.919433600Z
2019-09-02T19:57:15.886807400Z
2019-09-02T19:57:15.851085400Z
2019-09-02T19:57:15.816373600Z   <-- 10mins later as expected
2019-09-02T19:46:55.958533500Z
2019-09-02T19:46:55.937718900Z
2019-09-02T19:46:55.916665800Z
2019-09-02T19:46:55.894055800Z
2019-09-02T19:46:55.870170600Z
2019-09-02T19:36:53.224354300Z
...

sensemonster, 2019-09-03:
[image]

tvoverbeek, 2019-09-03:
@sensemonster How do you set the next wakeup? Looks like you schedule it at the end of the run.
In your log the jump from 19:46 to 19:58 is between 10 and 11 minutes.
The 1 hour jump is just when 10 minutes later is in the next hour, which is a bit suspect.

sensemonster, 2019-09-03:
I will copy paste an extract from my script but yes it is set right at the
end of the script.

sensemonster, 2019-09-03:
Attached is the bottom section of my script
[PiJuiceDataLoggerExtract.txt](https://github.com/PiSupply/PiJuice/files/3571018/PiJuiceDataLoggerExtract.txt)

tvoverbeek, 2019-09-03:
You set the time of the next wakeup at the beginning of the script.
Since you are using a cloud service the RPi obviously has an internet connection.
At boot systemd starts the timesyncd daemon. It will set the system time from the internet and copy it to the RTC and clear the wakeup flag soon after boot.
This might interfere with your time of next wakeup calculation.
To avoid any interference from timesyncd you could change the id-eeprom address to 52.
Then you will only use the RTC as clock source.
I would move the setting of the next alarm to the end of the script after checking that the time of next wake-up has not been passed already.

sensemonster, 2019-09-04:
Yes, I think I get it now.  It appears that the 10 minute incrememt should be to whatever the actual time value has been written to the hwclock just at the point of issuing the power off.

sensemonster, 2019-09-05:
I have changed the code as above suggestion and so far it is looking OK but will monitor until this evening - approximately 24hrs since my modification.

sensemonster, 2019-09-06:
Much better plot as per attached picture - showing the gradual cooling overnight temperature as expected. 
[image]

sensemonster, 2019-09-06:
Thanks again to @troverbeek for his patience and helpful guidance.

closed 2019-09-06 by sensemonster