PiSupply/PiJuice #371: Issues booting with Raspberry Pi 4

icsy7867, 2019-09-03:
I am trying to get the pijuice to work with my new RPI4.

I have bought a new pijuice and it is running the 1.3 firmware.  I have put dtoverlay=i2c-bcm2708 into my /boot/config.txt

as well as enabled i2c using raspi-config.

Everything worked fine the first time.  I set the shutdown setting on power loss, tested it and the pi shutdown.  However upon re-inserting power things have gotten a little weird.  The device seems to be in a bootloop.  my screen will flash every 2-3 minutes like it is trying to bootup (With the mutli-colored splash screen for an instant).  The pijuice is flashing blue/orange as expected.

Upon removing the pijuice board, the rpi4 boots without issue.  I have tried powering through the micro-USB connector on the pijuice as well as directly using the USB-C connector on the rpi4.

Any ideas?

ChristopherRush, 2019-09-03:
I would suggest to reset the HAT configuration back to the default
settings. You can do this by holding down SW1 and SW2 for 20 seconds.

icsy7867, 2019-09-03:
Thanks for the quick reply!

I went ahead and just reflashed my SD card.  Same behavior plugged in using micro-usb into the pijuice.

Seems to be in a bootloop.  I see the raspberry pi 4 activity light.  However I did catch a glance as a black screen with a small lightning bolt on the top left.  Could this mean the pi isnt getting enough power?  I could understand the little BP7X not supplying enough, but if I am plugged in using the micro USB cable, it should be sufficient correct?

tvoverbeek, 2019-09-03:
Seems you did not install the pijuice-base package or had a bad install.
The pijuice-base package installation does enable i2c and also enables the pijuice.service.
I do not know about the cause of the occasional "Data Corrupted" since you have the i2c-bcm2708 module enabled.
What kernel are you running (output of 'uname -a')?
Most recent is 4.19.66 from Aug 15.

icsy7867, 2019-09-03:
appreciate the response.

uname -a issues:

> Linux Raspberry[o 4.19.66-vl7

apt-get install pijuice-base tells me it is already installed (I installed pijuice-gui)

tvoverbeek, 2019-09-03:
Seems you are up-to-date. Just wondering how the pijuice service (the system task) got messed up.

icsy7867, 2019-09-03:
It seems to  be working other than sometimes saying "Data Corrupted" for the "Battery" field.  Could it be a wonky battery?

daphatty, 2019-09-05:
> It seems to be working other than sometimes saying "Data Corrupted" for the "Battery" field. Could it be a wonky battery?

Any chance you have a Pimoroni FanShim running in between the Pi and PiJuice? I saw similar "Data Corrupted" behavior with this configuration.

icsy7867, 2019-09-05:
Interestingly enough I had a 5v fan on the 5v and gnd pins.  It did seem to go the corrupted route more often.  

I also have a pimoroni fan shim on the way.  I'll definitely see if I notice when I install it

daphatty, 2019-09-05:
I've had a similar experience with other fans as well. I know the pijuice "works" with the Pi4 but there is definitely some additional engineering that needs to go into the pijuice so it "works properly" with the Pi4.

icsy7867, 2019-09-06:
So, I just got my pimoroni fan shim.

I have not noticed any data corrupt errors yet.  However I do get random reboots as the PiJuice thinks the power gets disconnected.

However, this does not seem to happen when plugged into the microUSB port on the pijuice itself.

hugo-a-garcia, 2019-10-03:
Hi

Just following up since I do have a Pi4 and the shim. Is it working properly? Also,did you stack them? Maybe someone can attach a picture? Please.

icsy7867, 2019-10-18:
> Hi
> 
> Just following up since I do have a Pi4 and the shim. Is it working properly? Also,did you stack them? Maybe someone can attach a picture? Please.

Here you go:
https://imgur.com/a/TkG6HjT

The pijuice is not tall enough to fit the pimoroni fan shim.  I had to use the raspberry pi 4 hightop case:
https://www.pishop.us/product/highpi-raspberry-pi-case-for-pi4/

And it still bulges slightly.  However it works and the fan kicks on and keeps everything fairly cool.

shawaj, 2019-10-18:
Could this potentially be related to #397 and #177 ? Maybe just a coincidence but seems some similarities in behaviour

prenticem3, 2019-11-15:
I too am receiving “Data Corrupted” errors on my Pi4. I have a Fanshim mounted below similar to the pics posted by the OP. In an earlier comments you asked if he had a Fanshim because you saw similar behavior with this configuration. 

Was there a resolution in that instance?

ChristopherRush, 2019-11-15:
Did you add the dtoverlay 'dtoverlay=i2c-bcm2708' to the '/boot config txt'
file?

prenticem3, 2019-11-15:
Right after I posted my reply I saw another post with this as a solution. Thank you very much for the fast reply. Will add this once I get home. 

Apologies for posting any not fully reviewing all solutions.

prenticem3, 2019-11-16:
Added the dtoverlay setting and it works like a charm. Thanks again!

tvoverbeek, 2019-11-16:
Since it works now, closing this issue.

closed 2019-11-16 by tvoverbeek