PiSupply/PiJuice #372: DATA_CORRUPTED and no version of firmware on new PiJuice

popecix, 2019-09-05:
Hello,

I just picked up a PiJuice Hat for a project I'm working on with an RPi3B+ (Buster w/ Xfce). It seems to be functioning, but in the Configuration menu the fields all show "DATA_CORRUPTED" and under firmware the version is showing as 0.0/unknown. The service appears to be loading fine, I'm seeing the address in i2cdetect, and I can poll the info from a python interpreter with GetStatus commands.

I've searched issues here and more broadly in google, but not finding anything similar to what I'm seeing. Potentially looking at deploying 40-50 of these in critical role for a live event production so want to understand (a) if this normal behavior, (b) if there is a problem and what the solution is, or (c) if it could just be a faulty unit.

Any insight greatly appreciated!

tvoverbeek, 2019-09-05:
What is the output of 'i2cdetect -y 1'. Expected is adresses 0x14 for the PiJuice and 0x68 for the (simulated) RTC. The entry for 0x68 should be 'UU' if the device-tree fragment from the HAT eeprom loaded correctly.
Since you wrote GetStatus works when using the python3 interpreter the I2C communication works.
My hunch is that it has something to do with running a different desktop on top of Buster lite.

popecix, 2019-09-10:
Apologies for the delay in following up on this, been a bit hectic with other project work.  I am  seeing the 0x14 and 0x68 in the i2cdetect.

I thought about the Xfce difference, but figured both it and PIXEL are just wrappers for X. Though I don't know enough about the inner workings of GTK+ vs. QT to be sure (my expertise is far more network infrastructure oriented, past Layer 4 it's usually someone else's problem...)

Have everything taken apart at the moment for another build, but as soon as I get it back together in the next day or two I'll run the GetFirmwareVersion and let you know.

tvoverbeek, 2019-09-10:
@popecix Just some additional info.
Pijuice_gui and pijuice_cli have to run as user pijuice. When installing the pijuice-base package
the pijuice user is created.
This is done by running a setuid wrapper which calls the python3 interpreter to run either pijuice_gui.py or pijuice_cli.py. The wrappers are pijuice_gui and pijuice_cli.
Also when started from a Desktop menu you need an entry in the sudoers file. See the github for details (https://github.com/PiSupply/PiJuice/blob/master/Software/Source/data/36x11-pijuice_xhost) and the postinst scripts for pijuice_base and pijuice_gui:
https://github.com/PiSupply/PiJuice/blob/master/Software/Source/debian-base/postinst
and
https://github.com/PiSupply/PiJuice/blob/master/Software/Source/debian-gui/postinst

ebarnard, 2019-09-18:
I'm seeing the exact same issue with an RPi 4B. This is with a clean Raspbian install and no other peripherals attached.
`i2cdetect -y 1` gives the expected output.

popecix, 2019-09-18:
@tvoverbeek hypothetically we could change, but functionally it's highly unlikely to switch everything over to SPI. I'll be checking today to see if 100K fixes the issue and if so, how it impacts the OLED functionality.

However, since these will be effectively headless units, and I've seen other posts stating that the PiJuice will 'work' as a UPS without any installation of software, what major issues would I have to simply not use the software? I.e. would the SW1/J5 still do the default startup/shutdown actions? Primarily just need hardware control (Momentary switching) of the unit to get it on and off appropriately.

Will get back with more when I look into 100k vs. 400k....

tvoverbeek, 2019-09-18:
@ebarnard On the RPi 4B the default i2c driver does not work with the PiJuice (probably because the i2c hardware in the SoC is different). Use the previous i2c driver by adding this line to/boot/config.txt: `dtoverlay=i2c-bcm2708` and reboot.
@popecix Waiting to find out how it goes with 100k clock.

popecix, 2019-09-18:
@tvoverbeek just finished testing, the default (100K) baud rate works with the PiJuice, but results in some noticeable screen tearing on OLED refresh. Not sure what options that leaves other than choosing one or the other to have working - in which case the OLED has priority.

Back to my other comment, if I'm not doing anything to call/interact with the PiJuice, simply using it as a buffer between changing external power sources, there wouldn't be any reason to need the GUI/service interaction and as such we could generally run at 400K and switch back to 100K as needed to do firmware updates?

Either way, we've found the root cause of the issue in my mind and this thread can be closed at your discretion.

Thanks for the help in getting to the bottom of this!

tvoverbeek, 2019-09-18:
@popecix Yes, using the PiJuice with default settings will work gine as UPS.
SW1 startup will work. Shutdown by SW1 requires cooperation of/communication with the RPi and will probably not work with 400k I2C clock.
There might be another solution if you do some rerouting of the PiJuice I2C connections.
Use the normal I2C at 400k for your OLED.
You can define a bit-banged I2C bus on 2 other GPIOs using the i2c-gpio overlay.
This will create a separate i2c bus (currently 3, but will be 10 in the future to be sure not to interfere with the extra i2c buses of the RPi4).
You will have to update the bus number from 1 to 3 (or 10) in the PiJuice software.
The bit-banged GPIOs will have to be connected to the normal I2C pins of the PiJuice (hence direct stacking cannot be used).

popecix, 2019-09-26:
@tvoverbeek thanks for the extra workaround idea, very interesting though, but since there will be 40-50 of these units deployed that would add too much time/complexity to the manufacturing per unit. We're still discussing internally how impactful the 100K rate is with the OLEDs vs additional functionality needs for the PiJuice. I'm using another momentary switch for power on/off already, so  it will probably end up being just using the PiJuice as a simple UPS.

Greatly appreciate the support and going to close out this issue. Thanks again!

closed 2019-09-26 by popecix

lowellbert, 2019-11-08:
> @ebarnard On the RPi 4B the default i2c driver does not work with the PiJuice (probably because the i2c hardware in the SoC is different). Use the previous i2c driver by adding this line [...]

Thanks this fixed similar issue, Using RP4 with firmware 1.3. Cloned drive and was working with RP3B+. I am assuming this issue is from using the RP4, any idea when new software will fix this issue?

tvoverbeek, 2019-11-09:
@lowellbert We are testing a fix. The firmware in the next release will not need the i2c-bcm2708 work-around for the Pi4.
I do not know when it will be available