PiSupply/PiJuice #374: PUSH SWITCH BUTTON

LuvSachdevaRFID, 2019-09-12:
I am looking to connect an external switch to the pijuice so that I can externally control the function that switch is doing.
Any recommendations for the switch I can use?
Also, Will, it works if I just solder the switch terminals underneath the switch button on the pijuice?
I will appreciate if anybody can help me out in this.

Thanks in advance.

LuvSachdevaRFID, 2019-09-20:
Thanks. It works great.
Is it possible to connect external LED's to get to know the battery status while charging?

tvoverbeek, 2019-09-21:
Not directly, but ...
What you can do is connect your external LED(s) to either the 2 available IO pins on PiJuice connector P3 and/or to any available GPIO on the RPi. In both cases you need to run a program on the Pi which samples the battery level and sets the external LEDs accordingly.

shawaj, 2020-10-24:
Closing this issue now. @LuvSachdevaRFID if you disagree feel free to reply

closed 2020-10-24 by shawaj