PiSupply/PiJuice #375: CLI Software for the PiJuice not working (hardware looks OK)

WenceslaoGrillo, 2019-09-15:
Hello!
I've just purchased an RPi 4 and a PiJuice, connected them together, and to the power. No monitor, no keyboard, no mouse. I set the RPi to find my Wifi and enabled SSH. Started it, could start a virtual VNC desktop with the user Pi, and another with another user. Successfully installed the software for the PiJuice with "sudo apt-get install pijuice-gui" (also installed a couple of dependencies), rebooted just in case. Now I have the icon for the charge level in the system tray. Right click, about says:
Software version: 1.5
Firmware version: 15.15
OS version: Linux 4.19.66-v7l+ #1253 SMP Thu Aug 15 12:02:08 BST 2019

The GUI configuration program shows "Battery: DATA_CORRUPTED", but otherwise looks normal to my newbie eyes. But I suspect this is part of a bigger problem.

Connected via SSH with the Pi user (directly with that user, not using "sudo su -") I try "pijuice_cli", the program comes up but if I try the "Status" button I get:

Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 2007, in <module>
    loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 286, in run
    self._run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 384, in _run
    self.event_loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 788, in run
    self._loop()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 825, in _loop
    self._watch_files[fd]()
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 404, in <lambda>
    event_loop, callback, self.get_available_raw_input())
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 502, in parse_input
    callback(processed, processed_codes)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 411, in _update
    self.process_input(keys)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 511, in process_input
    k = self._topmost_widget.keypress(self.screen_size, k)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 595, in keypress
    *self.calculate_padding_filler(size, True)), key)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 1590, in keypress
    key = self.focus.keypress(tsize, key)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 2271, in keypress
    key = w.keypress((mc,) + size[1:], key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/listbox.py", line 999, in keypress
    key = focus_widget.keypress((maxcol,),key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/wimp.py", line 540, in keypress
    self._emit('click')
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 460, in _emit
    signals.emit_signal(self, name, self, *args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 265, in emit
    result |= self._call_callback(callback, user_arg, user_args, args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 295, in _call_callback
    return bool(callback(*args_to_pass))
  File "/usr/bin/pijuice_cli.py", line 1907, in item_chosen
    callback()
  File "/usr/bin/pijuice_cli.py", line 191, in __init__
    self.main()
  File "/usr/bin/pijuice_cli.py", line 251, in main
    status_args = self.get_status()
  File "/usr/bin/pijuice_cli.py", line 206, in get_status
    general_info += ", %3d°C, %s" % (temp, status['battery'])
KeyError: 'battery'

The "General" button says: "Unable to connect to device"
The "Battery profile" button gives another traceback. It's last line is:
AttributeError: 'BatteryProfileTab' object has no attribute 'ext_profile_data'
And similarly, most other buttons just give unhandled exceptions.

I tried "i2cdetect -y 1" because I saw it was recommended in similar cases. I got this:

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

I think instead of 68 I should have got UU, but I don't know what to do about it.

I also read about doing some tests directly from the python 3 interpreter, but the import fails:

$ python3
Python 3.7.3 (default, Apr  3 2019, 05:39:12)
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pijuice import Pijuice
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'Pijuice' from 'pijuice' (/usr/lib/python3/dist-packages/pijuice.py)

However, if I disconnect the charger, the RPi goes on running without any interruption of the SSH connections via WiFi.

Any help will be much appreciated.

    Wences

ChristopherRush, 2019-09-15:
> File "", line 1, in

If you check the GitHub issues already, I think its issue #327 you will see
that there is a quick fix solution until we update the software.

shawaj, 2019-09-15:
You can also look here https://github.com/raspberrypi/linux/issues/3064#issuecomment-510057892

shawaj, 2019-09-15:
@ChristopherRush @tvoverbeek do we know yet whether this is something that RasPi will fix on their end or if it is something we need to do?

WenceslaoGrillo, 2019-09-15:
Awesome! You guys rock!
I did as #327 says and now everything is working! Thanks!

closed 2019-09-15 by WenceslaoGrillo

gatuno1, 2019-10-15:
I had the same error on RPi B+. running in `$ lsmod | grep i2c` I saw loaded both:
i2c_bcm2835            16384  0
i2c_bcm2708            16384  0
i2c_dev                16384  2

tishanm-github, 2020-01-28:
> If you check the GitHub issues already, I think its issue #327 you will see that there is a quick fix solution until we update the software.
> […](#)

Any idea when this will be fixed? I applied the change described in #327 and that fixed things with PiJuice but it broke my I2C sensors that were being connected via my Grove Base Hat. Without a fix, I can't use PiJuice which means the 6 I bought are useless to me now. Please help.

tvoverbeek, 2020-01-28:
@tishanm-github I am close to building the packages for the next release (1.6). This will use the bcm2835 i2c driver again. No need for the #327 work-around any more.

However, I was looking through the grove software and this is still based on stretch.
E.g. they use python3.5 and buster is on python3.7. Normally this should not matter.
The Grove I2C ports are connected directly to the Pi I2C bus.
The Grove HAT uses the same MCU as the PiJuice.
In their code they use the smbus2 module while PiJuice uses the original smbus module. Do not know if it really works if you use both at the same time (device contention?).
On all the newer Pis (Pi2,3,3+,4) we have seen the i2c clock varying depending on the CPU load.
On a lightly loaded system you get about half the specified clock speed. When the load increases the speed goes to the nominal 100kHz. Do not know how your sensor handles varying clock speeds.
Does your sensor work fine on the Pi4+GroveHAT without the PiJuice even on a varying CPU load?
A simple way to load 1 core at 100% is to use the following command: `cat /dev/urandom > /dev/null &`.

tishanm-github, 2020-01-30:
@tvoverbeek It appears that one of .our sensors was faulty. Right now everything except the PiJuice CLI works while using the default I2C module. We can live with that for now. Thanks for the quick response.