PiSupply/PiJuice #376: Run script when Charging Power present

ericjackson-eng, 2019-09-17:
I have a project where the PiJuice needs to turn off the USB ports when the Charging Power is removed and turn them back on when the Charging Power returns.  I have scripts to run that will handle the turning off and on of the ports and can have the "unbind" one start using the 'No Power' System Event (that works just fine).
Is there a System Event or way to run a particular script when the Charging Power returns?  I know the PiJuice detects this but cannot find an option to start a script when it is detected.
Thank you in advance for your help.

tvoverbeek, 2019-09-17:
@ericjackson-eng You will have to run your own script polling the PiJuice to detect when the charging input returns. Check the pijuice.status.GetStatus() function in the I2C API section at the end of the software README (https://github.com/PiSupply/PiJuice/tree/master/Software#i2c-command-api)
Check the battery status if it contains 'CHARGING_FROM_IN' or 'CHARGING_FROM_5V_IO'.

ericjackson-eng, 2019-09-18:
That worked great. Thank you so much for your help.

closed 2019-09-18 by ericjackson-eng