PiSupply/PiJuice #378: wakeup alarm not working

jsec83, 2019-09-22:
Hi,

I've got troubles when using the wakeup alarm, it won't work.
I've used the python script mentioned here: [https://github.com/PiSupply/PiJuice/issues/91#issuecomment-381324352](url)
I want the script to be launch every day at the same hour so I've changed the a['minute'] to 0 and a['hour'] from 'EVERY_HOUR' to a specific hour 12 for example (I know it is UTC time so in my timezone, it corresponds to 14:00).

Unfortunately, nothing happened at 14:00, even after changing the eeprom address to 52.
I've put the script in a crontab @reboot, i've used the "wakeup_enable.py" script and also put it in a crontab @reboot.

I've put a `sudo hwclock -s` in the /etc/rc.local.

What do I forget?

Thanks in advance

tvoverbeek, 2019-09-22:
No need to use wakeup_enable.py since the equivalent is already in https://github.com/PiSupply/PiJuice/issues/91#issuecomment-381324352
Also no `hwclock -s` in /etc/rc.local is needed since this is also in the script.

What could be is that we changed from python2 to python3.
So the crontab line should be `@reboot /usr/bin/python3 /home/pi/test.py` (not /usr/bin/python).
Have you tried to run your script directly to test if no errors are reported? 
For testing you could delete/comment out the last 2 lines of the script, so the Pi does not shut down. You can then check if it did what it whas supposed to do.

jsec83, 2019-09-22:
Thanks for the information concerning the `hwclock -s` and the wakeup_enable.py, yes the script is working well when I launch it alone directly in the terminal.
Off course, I've written `/usr/bin/python3` in the crontab since it is a python3 script.
I really think that the pijuice doesn't wake up. I' don't know why because I checked using pijuice_cli that the wakeup alarm is consistent with the alarm I set in the python script (just before the script shutdown the raspberry pi).

Any idea of test I could perform to find the problem?

jsec83, 2019-09-24:
I've tested the script with a wakeup alarm set to 'EVERY_HOUR' and the pi isn't waking up...
I've attached the script, do you have an idea of the problem?
The script is included in the crontab using this line:
`@reboot /usr/bin/python3 /home/pi/send_pwr_by_email.py`

[send_pwr_by_email.py.zip](https://github.com/PiSupply/PiJuice/files/3649627/send_pwr_by_email.py.zip)

ChristopherRush, 2019-10-01:
From what I can gather after a bit of testing with your script, the issue is with the log file. Must have something to do with permissions. Comment out the log file part of your script and see if it works ok, just to be sure its not something else.

tvoverbeek, 2019-10-02:
I tried to sent an answer earlier with a correcter working script, but somehow it did not make it to github. Now I am away from home and do not have access to my RPi with the working script.
From my memory:
Move the SetWakeup(True) after the 60 seconds sleep to just before the shutdown.  This to avoid timesyncd resetting the wakeup flag during the 60 seconds sleep.
Also add a call to ClearAlarmflag() just before the shutdown. The flag gets set when the alarm triggers but only if previously cleared.
With these changes I got your script to send me a  email every hour for at least 24 hours (after which I stopped the test)
For debugging you can add print statements at various places in your script and modify your crontab entry to:
@reboot /usr/bin/puthon3 /home/pi/send_power_by_email.py >> /home/pi/send_power_by_email.log 2>&1
Any output including error messages will appear I  the log file.
Hope I did not make any mistakes. Will be back home on Saturday Oct 5.

jsec83, 2019-10-02:
Thanks a lot @tvoverbeek, your solution works perfectly well!

closed 2019-10-02 by jsec83