PiSupply/PiJuice #379: Unable to change battery profile: ValueError: 'UNKNOWN' is not in list

mschuler, 2019-09-23:
I wanted to change the battery profile with FW version 1.2 but was unable to select my 12000mAh LiPo battery. I updated the firmware then to 1.3 but since then i get an error when opening the battery profile page in pijuice_cli:

```
pi@raspberrypi:~/pijuice $ pijuice_cli
Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 2007, in <module>
    loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 286, in run
    self._run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 384, in _run
    self.event_loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 788, in run
    self._loop()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 825, in _loop
    self._watch_files[fd]()
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 404, in <lambda>
    event_loop, callback, self.get_available_raw_input())
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 502, in parse_input
    callback(processed, processed_codes)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 411, in _update
    self.process_input(keys)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 511, in process_input
    k = self._topmost_widget.keypress(self.screen_size, k)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 595, in keypress
    *self.calculate_padding_filler(size, True)), key)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 1590, in keypress
    key = self.focus.keypress(tsize, key)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 2271, in keypress
    key = w.keypress((mc,) + size[1:], key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/listbox.py", line 999, in keypress
    key = focus_widget.keypress((maxcol,),key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/wimp.py", line 540, in keypress
    self._emit('click')
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 460, in _emit
    signals.emit_signal(self, name, self, *args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 265, in emit
    result |= self._call_callback(callback, user_arg, user_args, args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 295, in _call_callback
    return bool(callback(*args_to_pass))
  File "/usr/bin/pijuice_cli.py", line 1907, in item_chosen
    callback()
  File "/usr/bin/pijuice_cli.py", line 1003, in __init__
    self.refresh()
  File "/usr/bin/pijuice_cli.py", line 1123, in refresh
    self._read_chemistry()
  File "/usr/bin/pijuice_cli.py", line 1181, in _read_chemistry
    self.chemistries_idx = self.CHEMISTRY_OPTIONS.index(self.ext_profile_data['chemistry'])
ValueError: 'UNKNOWN' is not in list
```

I removed pijuice_base and installed it again without success. The error message remains. Is there a possibility to change the corrupt setting with a python script or similar?

mschuler, 2019-09-23:
I fixed the error this way:

```
pi@raspberrypi:~/pijuice $ python3
Python 3.7.3 (default, Apr  3 2019, 05:39:12)
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pijuice import PiJuice, pijuice_hard_functions, pijuice_sys_functions, pijuice_user_functions
>>> pijuice = PiJuice(1, 0x14)
>>> pijuice.config.SelectBatteryProfiles(0x13)
>>> pijuice.config.batteryProfiles
['BP6X_1400', 'BP7X_1820', 'SNN5843_2300', 'PJLIPO_12000', 'PJLIPO_5000', 'PJBP7X_1600', 'PJSNN5843_1300', 'PJZERO_1200', 'PJZERO_1000', 'PJLIPO_600', 'PJLIPO_500']
>>> pijuice.config.SetBatteryProfile('PJLIPO_12000')
{'error': 'NO_ERROR'}
```

pijuice_cli is working now without errors.

closed 2019-09-23 by mschuler

tvoverbeek, 2019-09-24:
Thanks for the report. I am aware of the problem and will try so fix it so pijuice-cli does not crash with an unknown profile but gives an error message instead.

logan893, 2022-06-22:
> Thanks for the report. I am aware of the problem and will try so fix it so pijuice-cli does not crash with an unknown profile but gives an error message instead.

Just commenting to say that pijuice-cli still crashes when the "chemistry" option is missing, and this workaround is still necessary in that scenario.