PiSupply/PiJuice #380: System switch resets on startup

ChristopherRush, 2019-09-25:
From a customer:
Recently purchased a PiJuice, so far so good. Am trying to power a small touch screen as well as the Pi. Am trying to use the system switch voltage pins to do so. How can I permanently set the system switch to 2100mA so on power-up voltage is present on those pins. Currently on startup it's always off.

*I have tested this scenario and was unable to replicate it on my setup.

Is there any reason which may cause this?

tvoverbeek, 2019-09-25:
What I can deduce from the firmware source is that the system switch state is not saved in NV memory. So if the MCU loses power the system switch state reverts to its power up state.
Tested it by shutting down, removing power, removing battery, insert battery, power up:
system switch state reverts to off

ChristopherRush, 2019-10-23:
Thanks, will document this.

shawaj, 2019-10-23:
@tvoverbeek @ChristopherRush is this something you think we should change / fix? Any reason for it not to be saved?

tvoverbeek, 2019-10-26:
System switch switches VSYS which is used to power external things. Maybe after a forced shutdown (e.g. due to low battery and/or no power) you do not want to power the external stuff immediately after power returns, but switch it on after checking the reason for the shutdown.
Therefore suggest to add an option to save the setting to NV memory which the user can select.