PiSupply/PiJuice #382: Pi does not get PiJuice time

rascalos, 2019-09-26:
Hello,
I try to use PiJuice on Pi 4.
All seems ok after /boot/config.txt modification (about dtoverlay=i2c-bcm2708 to DATA CORRUPTED issue), but I have a drift between Pi 4 time and PiJuice time. 
I can see it after each stop & start.
any ideas ?

rascalos, 2019-09-26:
To be more precise, the date of Pijuice is good but that of PI4 seems not to be updated.
Thanks.

tvoverbeek, 2019-09-26:
Does your Pi have a connection to the internet?
There is no automatic copying of the PiJuice RTC time to the RPi.
Without an internet connection the RPi time starts where it left off at previous shutdown,
With an internet connection the timesyncd daemon copies the time from the internet to the RPi time and also copies it to the RTC.
Can you explain your setup (internet yes/no, when do you check for the time difference)?

rascalos, 2019-09-26:
I'm sorry, my installation procedure sheet was not complete. "hwclock -s" command was missing in /etc/rc.local file.
Thank-you for your prompt response.
Regards,

closed 2019-09-26 by rascalos

tvoverbeek, 2019-09-26:
Glad you got it working