PiSupply/PiJuice #383: Battery is not charging upto 100% / swapping temp during charging

rene54321, 2019-09-28:
Hi,

I have received your wonderful piece of hw last week. I have charged the battery with the Pi turned off over the night but the piJuice is still charging ( indicated by the blue, green flashing LED ). This morning I have turned on the Pi to check the battery level with usb power connected. And it was 97% not 100%. Is this normal? 

Next problem is the battery temp this one is swapping between something around 40°C and above 100°C for a short time and is shows shortly Fault: charging temperature SUSPEND. Is my battery defect?

best regards René

tvoverbeek, 2019-09-28:
97% is fine. It might take a few battery cycles before you could see 100%

The outlier temperatures (>100°C) have been reported by other people.
The MCU software however believes them and suspends charging (battery too hot).
Could be a MCU software problem or something like a bad connection to the NTC resistor inside
the battery.

tvoverbeek, 2019-12-30:
The fluctuating temerature issue is a firmware problem. Solved in the upcoming firmware version (1.4).
Closing the issue.

closed 2019-12-30 by tvoverbeek