PiSupply/PiJuice #385: Issue with the installation of software

Arcade-Anima, 2019-09-30:
I'm a newb when it comes to linux and I've run into an issue where I cannot install the software for the pijuice.  I'm running Rasbian Strech on a RPI 3B.  When I go to type in any of the automatic installation options I get Unable to locate package pijuice-gui or base.  Neither works.  Any help would be useful.

ChristopherRush, 2019-10-01:
Do you have an internet connection? Try to run `sudo apt update` first.

norm-pi, 2019-10-06:
I have the same problem. I have run sudoku apt update first but I still get "Unable to locate package pijuice-base".

tvoverbeek, 2019-10-06:
@Arcade-Anima @norm-pi What version of Raspbian (Wheezy, Stretch, Buster) are you running.
What is the contents ov /etc/apt/sources/list and /etc/apt/sources.list.d/raspi.list?
Mine running Buster are:

/etc/apt/sources/list:
```
deb http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi
```

/etc/apt/sources.list.d:
```
deb http://archive.raspberrypi.org/debian/ buster main
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspberrypi.org/debian/ buster main
```
The pijuice packages are in the archive.raspberrypi.org repository.

ChristopherRush, 2019-11-10:
@norm-pi Do you still need help with this?

norm-pi, 2019-11-12:
Hello,

I thought I posted my success but thanks for asking. I found the trouble was OSMC default Python was 2.7, when I changed it 3.x I was able to install PiJuice successfully.

regards,

Norm

shawaj, 2020-02-22:
@Arcade-Anima have you managed to get things working now?

norm-pi, 2020-02-23:
Yes works fine once I set the default Python to 3.x.

shawaj, 2020-02-23:
@norm-pi thanks, I saw your comment above.

It was @Arcade-Anima that had opened the issue that I was wondering if they had fixed it?

I'm going to assume they have and close the issue.

If @Arcade-Anima had any further issues please post back here or open a new issue 👌

closed 2020-02-23 by shawaj