PiSupply/PiJuice #386: RTC fails to initialize when started from (low) battery

dandavson, 2019-10-02:
This took me a long time to get to the bottom of (i.e. that it's battery related) so sharing my understanding. I can reliably reproduce with the BP7X battery when it's down to about 8% or below. The rtc module fails to load (can be seen with dmesg, or by reading hwclock).

```dmesg | grep rtc```
returns:
```
[    4.171639] hctosys: unable to open rtc device (rtc0)
[    5.325332] rtc-ds1307: probe of 1-0068 failed with error -121
```
```sudo hwclock -r```
returns:
```
hwclock: Cannot access the Hardware Clock via any known method.
hwclock: Use the --debug option to see the details of our search for an access method.
```
This is the failed i2c transaction (first transaction to 0x68 after power up)
[image]

When the device is powered from 5V this issue does not present itself.

``` sudo hwclock -r```
returns
```2019-10-02 17:13:52.083064+0000```

```dmesg | grep rtc```
returns
```
[    4.263087] hctosys: unable to open rtc device (rtc0)
[    4.798585] rtc-ds1307 1-0068: registered as rtc0
```
The normal first i2c transaction to the hwclock (i.e. when powered from 5V)
[image]

I believe this may be related to: https://github.com/PiSupply/PiJuice/issues/273

dandavson, 2019-10-03:
I'm using a Pi3 B+, with a piJuice v2.2.2, fw version 1.3. Tested with Ubuntu Core 18.04 (arm64) and with DietPi ARMv6-Stretch (9.11)

dandavson, 2019-10-03:
Further to this, I have now seen the issue with the battery at 29% according to pi-cli, and the device powered via the rasperry pi (from a 2.4A supply). I have no peripherals/headers apart from the PiJuice (and HDMI and a USB dongle for KB/Mouse)

tvoverbeek, 2019-10-05:
I have seen the "RTC not recognizes at boot" in the following scenario:
-RPI not powered and PiJuice in low power mode (slow weak blink of LED D1).
-Boot the PI by connecting power to the Pi.
Problem is then the PiJuice MCU is waking up too late to respond to the RTC initialization by the kernel (especially on the much faster Pi4).

jsec83, 2019-10-11:
I've gor the same problem.
I try to wake up my raspberry Pi with the PiJuice and only the battery (no other power source plugged). 
The battery is charged at 89% and I have the same issue: the RTC is not regognized at boot so impossible to synchronize the system clock with the hwclock ...
My project needs to work only on battery (without any other power source). How can I "slow down" the raspberry pi boot in order to make the RTC to be recognized?

Thanks in advance.

tvoverbeek, 2019-10-11:
After boot you could run a script which does the following:
- Check if /dev/rtc exists.
- If so, exit
- If not `sudo rmmod i2c-ds1307` followed by `sudo modprobe i2c:ds1339` or `sudo modprobe i2c:ds1307`

Use i2c:ds1339 if you want to access the alarm function from linux, otherwise use i2c:ds1307

jsec83, 2019-10-11:
Hi tvoverbeek,

unfortunately, it dosn't work...

`$ sudo rmmod i2c-ds1307`
`rmmod: ERROR: Module i2c_ds1307 is not currently loaded`
and just after I typed:
`$ sudo modprobe i2c:ds1339`
Nothing happened, so I've tried:
`$ sudo hwclock -r`
`hwclock: Cannot access the Hardware Clock via any known method.`
`hwclock: Use the --debug option to see the details of our search for an access method.`

Do you have any idea?

Thanks in advance

tvoverbeek, 2019-10-11:
@jsec83 Need a bit more info. What is the pijuice eeprom address?
Anything specific in /boot/config.txt? Could you list the contents of /boot/config.txt?
Which Raspbian version (Stretch/Buster)?
Which Pi (e.g. Pi3b+, Pi4b, Pi-zeroW, ...)?

jsec83, 2019-10-11:
@tvoverbeek Thanks again for your answer.
Off-course, I send you all the information you need.

I use Raspbian Stretch on a Pi-zeroW with a classical PiJuice (not the PiJuice zero).
and my /boot/config.txt file is attached to this post.
[config.txt](https://github.com/PiSupply/PiJuice/files/3719531/config.txt)

dandavson, 2019-10-12:
> Problem is then the PiJuice MCU is waking up too late to respond to the RTC initialization by the kernel (especially on the much faster Pi4).

I think that the rtc initialization happens several seconds after power-on, so I find it hard to believe the PiJuice MCU takes that long to wake up? If so this is a serious issue with the design.

I'll try your rmod / modprobe work-around and report back.

tvoverbeek, 2019-10-12:
@jsec83 There are some suspect lines at the end of your config.txt.
Why do you enable the lirc-rpi overlay? DO you use an infra-red remote with your Pi-zeroW?
If not, delete it. By the way the lirc-rpi overlay is obsolete and has been replaced by gpio-ir.
See the README file in /boot/overlays.
Also the memory reserved for the GPU is on the high side. Do you use the Pi-ZeroW for heavy video use?
The most troubling one  is `dtoverlay=gpio-no-irq`. This will almost certainly cause the hardware I2C to fail. Delete it.
Was this SD card used in some other system?

jsec83, 2019-10-12:
@tvoverbeek 
Thanks for have a look to my config.txt.
No there is no infra-red remote, I will remove the lirc-rpi line
The high GPU memory is required because I use a camera with this raspberry pi. I think it is required to record video with the camera.
OK I will try to remove the line `dtoverlay=gpio-no-irq`, I'll keep you informed if it works.

jsec83, 2019-10-16:
@tvoverbeek OK after testing many times after doing all the modifications you mentioned about the /boot/config.txt, it appears that it doesn't work.
I wrote a program that wakes up the Pi zero every 5 min. After booting, the program wait 1 min (time.sleep) then write the battery charge into a file and then shutdown.
The program worked during 2 hours, and during this 2 hours, the pi zero woke up every 5 min and wrote in a file the battery charge level. But one time after 2 hours (I don't know why at this time), the program crashed (with this error: `hwclock: Cannot access the Hardware Clock via any known method.` `hwclock: Use the --debug option to see the details of our search for an access method`) and the pi doesn't shutdown and stayed awake. For information, the battery charge level at this time was 91%. So it is not due to low battery level...
Do you have an idea? For my project, I really need to have the wake up feature working properly...

Thanks in advance

tvoverbeek, 2019-10-17:
@jsec83 Could you show us your program? If you do not want to post it here send me an email (tvoverbeek@gmail.com). Also your current config.txt.
Is your Pi-ZeroW connected to the internet (since it has built-in wireless)?
Do you set the 5 minute periodic wakeup from the GUI/CLI? Or do you set the next alarm from your program?

jsec83, 2019-10-17:
@tvoverbeek OK I send you the program by email.
Concerning the other questions:
- yes the Pi-ZeroW is connected to internet and send data by ftp
- The 5 minute periodic wakeup is set using the python program by a wake up alarm. I am not using the GUI/CLI interface.

jsec83, 2019-10-18:
@tvoverbeek I forgot to send you the current /boot/config.txt, here it is.
[config.txt](https://github.com/PiSupply/PiJuice/files/3744689/config.txt)

ChristopherRush, 2019-11-09:
I'd just like to add here that my PiJuice Zero is also experiencing this issue. Sometimes the RTC does not initialise at boot. I have EEPROM address set to 0x52 and added RTC 1339 overlay to config.txt file. - Still trying to debug. Using latest Rapbian Lite on a Pi Zero W with battery connected 80+%.

tvoverbeek, 2019-11-10:
Here is a way to get it working.
- Put the HAT eeprom at 0x52, so the device-tree fragment is not loaded
  (Not needed for the PiJuice Zero since it does not have an eeprom).
- Wait till I2C works, i.e by trying pj.status.GetBatteryVoltage()
- Load the module with modprobe: `sudo modprobe i2c:ds1307` if you do not need to control
  the alarm function from Raspbian. Otherwise `sudo modprobe i2c:ds1339` which is also
  what the device-tree fragment in the eeprom does.
- At this point the module is loaded, but there is no /dev/rtc and `sudo hwclock -r` fails.
- Send this command (from a terminal or script):
  `echo ds1307 0x68 | sudo tee /sys/class/i2c-adapter/i2c-1/new_device` for ds1307 or
  `echo ds1339 0x68 | sudo tee /sys/class/i2c-adapter/i2c-1/new_device` for ds1339.
- Now you should have a /dev/rtc and `sudo hwclock -r` works.
@jsec83 tested this and it works for him.

ChristopherRush, 2019-11-10:
@tvoverbeek Thanks, will test tomorrow morning.

ChristopherRush, 2019-11-15:
@tvoverbeek I get the following error `modprobe: FATAL: Module i2c::ds1339 not found in directory /lib/modules/4.19.75+`

tvoverbeek, 2019-11-15:
@ChristopherRush only a single colon, not a double colon in i2c:ds1339

ChristopherRush, 2019-11-15:
@tvoverbeek Aah yes, must wear my glasses these days :)

ChristopherRush, 2019-11-15:
Here is the response I get `tee: /sys/class/i2c-adapter/i2c-1/new_device: Invalid argument`
[image]

tvoverbeek, 2019-11-15:
@ChristopherRush You also get the 'Invalid argument' if the device already exists.
Look at ls /sys/class/i2c-adapter/i2c-1.
If the device exists there ia a directory called 1-0068 (if there is only 1 i2c device).
You obviously have more.

ChristopherRush, 2019-11-15:
Yes device exits there, but still cannot access hwclock.

tvoverbeek, 2019-11-15:
Suggest to start over from a fresh boot.
Load the module (ds1307 or ds1339)
and then the echo etc.
After the echo you should have 'UU' in position 0x68 of the 'i2cdetect -y
1' output

ChristopherRush, 2019-11-16:
Oddly this only occurs when powering from the battery. If my device is in an isolated/remote location I cannot reboot it, it simply stays on and drains battery as it exits script with an error. Alternatively I could check to see if RTC is available if not then try to load it then reboot if all else fails.

tvoverbeek, 2019-11-16:
@ChristopherRush For this to work there should be no rtc overlay in /boot/config.txt. If there is the module is already loaded and the MCU is not ready yet when the module tries to initialize the RTC.

ChristopherRush, 2019-11-16:
@tvoverbeek Thanks, so far results are good. Pi boots up every 10mins for last 3 hours.

shawaj, 2020-12-05:
@tvoverbeek @ChristopherRush is there anything further we need to do here such as updating any software or firmware?

tvoverbeek, 2020-12-05:
@shawaj No, nothing to do as far as I am concerned.