PiSupply/PiJuice #387: Can't Install Pijuice-base under OSMC

norm-pi, 2019-10-06:
I am using the latest version of OSMC. When I use the following instructions the install fails with "Unable to locate package pijuice-base".

OSMC/Kodi Installations
When installing the pijuice-base package the setup procedure assumes that the default user, which is normally 'pi' on Raspbian, has the privilege to sudo to another user 'pijuice'. On Raspbain user 'pi' has this privilege by default. To overcome this you need to edit the sudoers file with the following:

sudo visudo
Add the following line:

osmc ALL=(pijuice) ALL

tvoverbeek, 2019-10-07:
@norm-pi These instructions apply only after you have installed the pijuice-base package.

SInce the pijuice packages are in an raspberry pi specific repository they are not available in OSMC.
So you have to install the package manually.
You will find it in https://github.com/PiSupply/PiJuice/blob/master/Software/Install/pijuice-base_1.5_all.deb
Before you install it make sure you have installed the packages it depends on:
```
python3
i2c-tools
python3-smbus
python3-urwid
```
All these are Debian packages (not raspberry pi specific) so they should be available in OSMC.
Then install the downloaded pijuice_base: `sudo dpkg -i pijuice-base_1.5_all.deb`
After this follow the instruction you referenced.

norm-pi, 2019-10-07:
Thanks for your help so far.

I have everything installed but when I run pijuice_cli I get the PiJuice CLI window stating "No PiJuice found or Failed to use I2C bus. Check if I2C is enabled". running sudo pijuice_cli gives the same results.

Reading package lists... Done
Building dependency tree       
Reading state information... Done
pijuice-base is already the newest version (1.5).
0 upgraded, 0 newly installed, 0 to remove and 30 not upgraded.

tvoverbeek, 2019-10-08:
2 things to check:
Does /dev/i2c-1 exist?
What is the output from `i2cdetect -y 1`
My guess is that i2c is not enabled in OSMC

norm-pi, 2019-10-08:
I installed the following;

python3
i2c-tools
python3-smbus
python3-urwid

Then installed the .deb file. After a reboot OSMC will not start, there is not even any activity on the ethernet LED's. I tried 1.5 first and it failed then 1.4 as shown below and it has failed also. I've had to reinstall OSMC.

_$ sudo dpkg -i pijuice-base_1.4_all.deb
Selecting previously unselected package pijuice-base.
(Reading database ... 27077 files and directories currently installed.)
Preparing to unpack pijuice-base_1.4_all.deb ...
Unpacking pijuice-base (1.4) ...
Setting up pijuice-base (1.4) ...
Creating user pijuice
Adding user pijuice to group i2c
Adding user osmc to group pijuice
settings file created
Setting ownership and permissions
Enabling I2C
/var/lib/dpkg/info/pijuice-base.postinst: line 41: lua: command not found
* Failed to read active DTB
pijuice.service is not active running, attempting to enable
Removed /etc/systemd/system/pijuice.service.
Created symlink /etc/systemd/system/multi-user.target.wants/pijuice.service → /lib/systemd/system/pijuice.service._

tvoverbeek, 2019-10-08:
The part of the pijuice-base post install script which enabled I2C support uses a lua script to edit config files.
Obviously lua is not installed on your OSMC system.
Can you try to install lua? On Raspbian it is done with `sudo apt-get install luajit`
After lua is installed try to install pijuice-base again.
Hopefully it should then complete without errors.

norm-pi, 2019-10-08:
No luck.

Re-installed OSMC and installed all the dependancies + luajit. Rebooted fine. Tried to install the pijuice deb file and it crashed again.

sudo dpkg -i pijuice-base_1.5_all.deb
Selecting previously unselected package pijuice-base.
(Reading database ... 27099 files and directories currently installed.)
Preparing to unpack pijuice-base_1.5_all.deb ...
Unpacking pijuice-base (1.5) ...
Setting up pijuice-base (1.5) ...
Creating user pijuice
Adding user pijuice to group i2c
Adding user osmc to group pijuice
settings file created
Setting ownership and permissions
Enabling I2C
/var/lib/dpkg/info/pijuice-base.postinst: line 44: lua: command not found
Failed to read active DTB
pijuice.service is not active running, attempting to enable
Created symlink /etc/systemd/system/multi-user.target.wants/pijuice.service → /lib/systemd/system/pijuice.service.

tvoverbeek, 2019-10-09:
@norm-pi My bad. I suggested the wrong lua package.
Could you try again with `sudo apt-get install lua5.1`?
After the install try the command `lua -v` to check if lua installed  properly.
Then (finally) the instal of pijuice-base should succeed.

norm-pi, 2019-10-09:
Some progress but not there yet. See below for installation of lua, installation of pijuice-base .deb file and running commands from pijuice import PiJuice & pj=PiJuice(1,0x14).

Although rebooting after installation no longer crashes OSMC I still get the PiJuice CLI window stating "No PiJuice found or Failed to use I2C bus. Check if I2C is enabled"

$ lua -v
Lua 5.1.5  Copyright (C) 1994-2012 Lua.org, PUC-Rio

sudo apt-get install -f
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Correcting dependencies... Done
The following additional packages will be installed:
  dh-python i2c-tools libgdbm3 libmpdec2 libperl5.24 libpython3-stdlib libpython3.5-minimal libpython3.5-stdlib perl perl-modules-5.24 python3 python3-minimal
  python3-smbus python3-urwid python3.5 python3.5-minimal
Suggested packages:
  libdpkg-perl libi2c-dev python-smbus perl-doc libterm-readline-gnu-perl | libterm-readline-perl-perl make python3-doc python3-tk python3-venv python3.5-venv
  python3.5-doc binutils binfmt-support
Recommended packages:
  read-edid rename
The following NEW packages will be installed:
  dh-python i2c-tools libgdbm3 libmpdec2 libperl5.24 libpython3-stdlib libpython3.5-minimal libpython3.5-stdlib perl perl-modules-5.24 python3 python3-minimal
  python3-smbus python3-urwid python3.5 python3.5-minimal
0 upgraded, 16 newly installed, 0 to remove and 30 not upgraded.
1 not fully installed or removed.
Need to get 10.6 MB of archives.
After this operation, 55.1 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://ftp.debian.org/debian stretch/main armhf libpython3.5-minimal armhf 3.5.3-1+deb9u1 [565 kB]
Get:2 http://ftp.debian.org/debian stretch/main armhf python3.5-minimal armhf 3.5.3-1+deb9u1 [1,441 kB]
Get:3 http://ftp.debian.org/debian stretch/main armhf python3-minimal armhf 3.5.3-1 [35.3 kB]                                                                  
Get:4 http://ftp.debian.org/debian stretch/main armhf libmpdec2 armhf 2.4.2-1 [69.6 kB]                                                                        
Get:5 http://ftp.debian.org/debian stretch/main armhf libpython3.5-stdlib armhf 3.5.3-1+deb9u1 [2,085 kB]                                                      
Get:6 http://ftp.debian.org/debian stretch/main armhf python3.5 armhf 3.5.3-1+deb9u1 [229 kB]                                                                  
Get:7 http://ftp.debian.org/debian stretch/main armhf libpython3-stdlib armhf 3.5.3-1 [18.6 kB]                                                                
Get:8 http://ftp.debian.org/debian stretch/main armhf dh-python all 2.20170125 [86.8 kB]                                                                       
Get:9 http://ftp.debian.org/debian stretch/main armhf python3 armhf 3.5.3-1 [21.6 kB]                                                                          
Get:10 http://ftp.debian.org/debian stretch/main armhf perl-modules-5.24 all 5.24.1-3+deb9u5 [2,722 kB]                                                        
Get:11 http://ftp.debian.org/debian stretch/main armhf libgdbm3 armhf 1.8.3-14 [28.5 kB]                                                                       
Get:12 http://ftp.debian.org/debian stretch/main armhf libperl5.24 armhf 5.24.1-3+deb9u5 [2,850 kB]                                                            
Get:13 http://ftp.debian.org/debian stretch/main armhf perl armhf 5.24.1-3+deb9u5 [219 kB]                                                                     
Get:14 http://ftp.debian.org/debian stretch/main armhf i2c-tools armhf 3.1.2-3 [57.6 kB]                                                                       
Get:15 http://ftp.debian.org/debian stretch/main armhf python3-smbus armhf 3.1.2-3 [9,716 B]                                                                   
Get:16 http://ftp.debian.org/debian stretch/main armhf python3-urwid armhf 1.3.1-2+b1 [169 kB]                                                                 
Fetched 10.6 MB in 30s (353 kB/s)                                                                                                                              
Selecting previously unselected package libpython3.5-minimal:armhf.
(Reading database ... 24275 files and directories currently installed.)
Preparing to unpack .../0-libpython3.5-minimal_3.5.3-1+deb9u1_armhf.deb ...
Unpacking libpython3.5-minimal:armhf (3.5.3-1+deb9u1) ...
Selecting previously unselected package python3.5-minimal.
Preparing to unpack .../1-python3.5-minimal_3.5.3-1+deb9u1_armhf.deb ...
Unpacking python3.5-minimal (3.5.3-1+deb9u1) ...
Selecting previously unselected package python3-minimal.
Preparing to unpack .../2-python3-minimal_3.5.3-1_armhf.deb ...
Unpacking python3-minimal (3.5.3-1) ...
Selecting previously unselected package libmpdec2:armhf.
Preparing to unpack .../3-libmpdec2_2.4.2-1_armhf.deb ...
Unpacking libmpdec2:armhf (2.4.2-1) ...
Selecting previously unselected package libpython3.5-stdlib:armhf.
Preparing to unpack .../4-libpython3.5-stdlib_3.5.3-1+deb9u1_armhf.deb ...
Unpacking libpython3.5-stdlib:armhf (3.5.3-1+deb9u1) ...
Selecting previously unselected package python3.5.
Preparing to unpack .../5-python3.5_3.5.3-1+deb9u1_armhf.deb ...
Unpacking python3.5 (3.5.3-1+deb9u1) ...
Selecting previously unselected package libpython3-stdlib:armhf.
Preparing to unpack .../6-libpython3-stdlib_3.5.3-1_armhf.deb ...
[... cut ...]
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pijuice import PiJuice
>>> pj=PiJuice(1,0x14)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/pijuice.py", line 1540, in __init__
    self.interface = PiJuiceInterface(bus, address)
  File "/usr/lib/python3/dist-packages/pijuice.py", line 24, in __init__
    self.i2cbus = SMBus(bus)
PermissionError: [Errno 13] Permission denied
>>>

tvoverbeek, 2019-10-09:
We are getting there (slowly).
It seems your default OSMC user has no access to the i2c bus.
What is the output from `ls -l /dev/i2c-1`?
On Raspbian it is:
```
pi@raspberrypi:~ $ ls -l /dev/i2c-1
crw-rw---- 1 root i2c 89, 1 Oct  9 15:26 /dev/i2c-1
pi@raspberrypi:~ $ 
```
On Raspbian if you belong to the i2c group, you have access.
If it is similar on OSMC add your default user to the i2c group, then (maybe) it works.

norm-pi, 2019-10-09:
The output from ls -l /dev/i2c-1 is the following;

$ ls -l /dev/i2c-1
crw-rw---- 1 root i2c 89, 1 Oct  9 15:47 /dev/i2c-1

so it is the same as your example output. The default user in OSMC is osmc, how do I add it to the i2c group?

norm-pi, 2019-10-09:
This is the last 3 lines in /etc/group

osmc:x:1000:
i2c:x:112:pijuice,osmc
pijuice:x:1001:osmc

tvoverbeek, 2019-10-09:
From your /etc/group the osmc user already belongs to the i2c group.
Does `i2cdetect -y 1` work?

norm-pi, 2019-10-09:
Success! Although I don't know why. I ran i2cdetect -y 1 as you suggested  and I got a positive response so tried pijuice_cli and it worked. Can't thank you enough for your assistance.

I have reprogrammed button 2  to SYS_FUNC_HALT_POW_OFF, but would also like to have the LED's on the pijuice turn off or dim to low level. How can I do this?

tvoverbeek, 2019-10-10:
Glad it works now.
If the Pi is fed by the PiJuice and there is no external power then with SYS_FUNC_HALT_POW_OFF the PiJuice removes the 5V to the Pi after 30 seconds and the PiJuice MCU goes to a low power state with faint very slowly blinking LEDs.
Otherwise you would have to use your own shutdown function which turns of the LEDs and turns the 5V off after xx seconds (to have sufficient time to complete the shutdown).

norm-pi, 2019-10-10:
Is there a function to turn off the LED's or do you use the function to set the brightness level to zero?

tvoverbeek, 2019-10-10:
Either set the RGB levels to 0, or set the led function to 'NOT_USED'.

norm-pi, 2019-10-10:
OK thanks again.

Where can I get documentation listing all the pijuice functions?

closed 2019-10-11 by norm-pi