PiSupply/PiJuice #388: Setting the System Clock & RTC using the API  (Python)

trondsen, 2019-10-07:
Hello,

Is it possible to control and read out status of the RTC, in a similar way  to “control” and “power”?
The readme file is lacking some explanation here. I have tested out the different functionalities of 
pijuice.status.x()    and pijuice.power.x()

When I for instance try:  pijuice.rtcAlarm.GetControlStatus() its returning the same data as expected when calling pijuice.status.GetStatus()

I assume full functionality control via API is not implemented yet? Is there another way to do this? 

Thanks 

Regards
Jonas

ChristopherRush, 2019-10-07:
`pijuice.rtcAlarm.GetControlStatus()` should return:
`{'data': {'alarm_wakeup_enabled': True, 'alarm_flag': False}, 'error': 'NO_ERROR'}`
and
`pijuice.rtcAlarm.GetAlarm()` should return:
`{'data': {'second': 0, 'minute': 0, 'hour': 'EVERY_HOUR', 'day': 'EVERY_DAY'}, 'error': 'NO_ERROR'}`
or similar (I have set it to wakeup every hour here).

tvoverbeek, 2019-10-07:
Yes, the readme file is missing quite a few definitions.

Best source is the code for the pijuice module itself:
https://github.com/PiSupply/PiJuice/blob/master/Software/Source/pijuice.py
You can read or set the RTC time, read or set the alarm and related flags.
For example right now my pj.rtcAlarm.GetTime() returns:
`{'data': {'second': 37, 'minute': 28, 'hour': 16, 'weekday': 2, 'day': 7, 'month': 10, 'year': 2019, 'subsecond': 0, 'daylightsaving': 'NONE', 'storeoperation': False}, 'error': 'NO_ERROR'}`
To set the time fill a dictionary formatted the same as the data returned by GetTIme:
pj.rtcAlarm.SetTIme(dateTime) where dateTime is the dictionary.

closed 2019-10-23 by ChristopherRush