PiSupply/PiJuice #389: Installation never works

Dieseljesus, 2019-10-08:
Hi, i´ve tried to install the Pijuice numerous times with little success (at least 5-6+ times). This is my setup:

HW: Raspberry Pi 3B+
OS: Tried latest Raspbian Buster, Kali, Arch Linux

Problem: Doing the installation of the Pijuice-Gui driver according to the guides. Most of the times i get the Pijuice app in the menu, but almost never the icon in the systray. When i click the app in the menu, i see the hourglass and then nothing happens.

I had success installing a few days ago though before i wiped the system. Then i got the settings app to start but it never saved any settings i changed.

I have done a Systemctl enable pijuice.service. This is kinda disturbing right now since i cant see how much power i have left and also i cant config the buttons which means i have to press SW1 for 20+ seconds every time i shutdown for the screen to turn off.... This is really doing my head in!

Any solutions to this?

tvoverbeek, 2019-10-09:
@Dieseljesus Suggest to stick with Raspbian Buster (since that is what I have on my Pis)
Does '/dev/i2c-1' exist?
Assuming you have i2c-tools installed (i2c-tools is a dependency of pijuice-base which in turn is a dependency of pijuice-gui) what is the output of `i2ctools -y 1` when the PiJuice is connected to the Pi?
Is pijuice-base installed?

Dieseljesus, 2019-10-09:
hmmm...i get the 'could not open file /dev/i2c-1: no such file or directory. Any tips to this?

i2c tools installed, i2c enabled in raspi-config. lsmod shows that i2c-dev is running and I have added the i2c lines to /etc/modules, the lines to boot config etc, but still that message?!

tvoverbeek, 2019-10-10:
I am wondering how you installed pijuice-base.
The pijuice-base post install script does all this: enable i2c, update /etc/modules etc.
Do you run under another userid (not pi, uid 1000)?
`ls -l /dev/i2c-1` should show read/write access for group i2c.
If your user does not belong to the i2c group you get permission errors.
The /dev/i2c-1 is created by the i2c-dev module.
Is there an entry for i2c1 or i2c-arm in /boot/config.txt (e.g. `dtparam=i2c-arm=on`)?

Dieseljesus, 2019-10-10:
Huge thanks for all help tvoverbeek. 

* I am using the default pi user of Raspbian Buster
* I did sudo apt-get install pijuice-base / pijuice-gui so both packages are installed according to the installation guide.
* dtparam=i2c-arm=on is added in config.txt

The rest i have to check up on when im home again later today. Once again, thanks for all help!

ollietb, 2019-10-10:
This is also happening for me

Dieseljesus, 2019-10-10:
@tvoverbeek Can't do ls -l /dev/i2c-1 because there is no file or directory.

The user is 'pi', the group is 1000 as well as a few others. Also if i do 'id pi' i see the group 1001 is pijuice. Any ideas regarding this?

tvoverbeek, 2019-10-11:
After boot could you look at the output of `dmesg` to see if there is anything i2c related?
Also, are there any other i2c busses enabled. Look at `ls -l /dev/i2c*`

Dieseljesus, 2019-10-11:
@tvoverbeek i found these while running dmesg:

pinctrl-bcm2385 3f200000.gpio: pin gpio2 already requested by leds; cannot claim for 3f804000.i2c
pinctrl-bcm2385 3f200000.gpio: pin-2 (3f804000.i2c) status -22
pinctrl-bcm2385 3f200000.gpio: Could not request pin 2 (gpio2) from group gpio2 on device pinctrl-bcm2385
i2c-bcm2385 3f804000.i2c: Error applying setting, reversing things back.
i2c-bcm2385: Probe of 3f804000.i2c failed with error -22

...something like that

Also the ls command shows that /dev/i2c-3 is open....

This is a bit advanced for my limited linux knowledge. Any ideas? thans once again!

tvoverbeek, 2019-10-11:
It seems you have some other peripheral connected to the GPIO which is using a pin needed by i2c.
The presence of /dev/i2c-3 means you have defined a bit-banged (software) i2c.
What else is connected?
With this configuration the PiJuice cannot be controlled by the pijuice software, which is why the installation fails.
Can you list your /boot/config.txt?

ollietb, 2019-10-12:
I have a hyperpixel screen attached to the hat which is probably the cause. So it’s not possible to use the pijuice software with a small screen?

tvoverbeek, 2019-10-12:
@ollietb The Hyperpixel screens use all GPIOs.
See e.g. https://pinout.xyz/pinout/hyperpixel.
The GPIOs for I2C (GPIO2 and GPIO3, pins 3 and 5) are used for V-Sync and H-Sync on the hyperpixel screens.
In other words the hyperpixel screens cannot be used with any other HAT.

If you do not need the touch interface on the hyperpixel then you could use bit-banged I2C for the Pijuice on the GPIOs reserved for touch. However this requires rerouting the I2C clock and data lines to pins 3 and 5 of the PiJuice.
In addition you need to modify the software to use the I2C bus assigned to the bit-banged version (used to be 3, will change to 10)

ollietb, 2019-10-14:
Ah that's a shame. Thank you for your time.

ollietb, 2019-11-04:
For anyone else who might come across this thread looking to use the PiJuice with a screen, I've tested with the [Adafruit PiTFT Plus 480x320 3.5" TFT+Touchscreen](https://thepihut.com/collections/raspberry-pi-screens/products/adafruit-pitft-plus-480x320-3-5-tft-touchscreen) and it works. The hi-res Hyperpixel screen does not work.

tvoverbeek, 2019-11-04:
Closing this issue since yu cannot use any other HAT with a hyperpixel screen.

closed 2019-11-04 by tvoverbeek