PiSupply/PiJuice #390: PiJuice software under different user

WenceslaoGrillo, 2019-10-09:
Hello!
Apologies if this doesn't belong here. If there is indeed an issue, it's probably a hardware problem but between the keyboard and the chair.
I'm trying to use my Rpi 4 with a different user, not the default almighty Pi. It's a headless box, so everything is SSH or VNC.
With the Pi user, the software works OK, both CLI and GUI, and I have the icon on the system tray of the desktop... everthing nice and dandy.
But with a different user I don't get the icon. In the menu, under Preferences, I do have "PiJuice Settings" but it doesn't do anything. I click it and nothing, not even a screen blink. And on CLI, the command pijuice_cli gives the usual "text mode GUI" (kinda ncurses) but with an error message: No PiJuice found or Failed to use I2C bus. Check if I2C is enabled.
I added this other user to the pijuice group in hopes that that would fix it, but no joy.

Is running the software under another user just not possible? Or is this something I can fix from the config? Ideally I would like both, Pi and the other user to be able to run the PiJuice software.

Thanks in advance for your help:

Wences

tvoverbeek, 2019-10-09:
It is possible to run the software under another user.
You already added the user to the pijuice group.
But you also need to add the user to the i2c group to get access to the i2c bus.
You can test with `i2cdetect -y 1` if i2c access works.

WenceslaoGrillo, 2019-10-09:
Hello, again! Thanks for the fast response.
I added the other user to the i2c group. Now I get the same output if I run i2cdetect -y 1 with that user or Pi, and I have the icon for the other user, but now Pi doesn't have the icon. Both users are able to run the GUI software, but Pi needs to go through the menu.
Frankly, I'm perfectly happy with that, but I thought I'd let you know in case it's some kind of unwanted thing.
Thanks again:

Wences

tvoverbeek, 2019-10-10:
@WenceslaoGrillo Thanks for the bug report about the tray icon.
I found the reason why it happens. The first user after boot gets the icon. Looking for a fix.
Closing the issue. If you do  not agree reopen it.

WenceslaoGrillo, 2019-10-10:
Thanks once again for your help. Yes, if you don't need the issue open to keep track yourself, we'll leave it closed. I'll keep an eye open for new versions with the bug fixed.

closed 2019-10-10 by WenceslaoGrillo

tvoverbeek, 2019-10-13:
@WenceslaoGrillo Fix in github. Commit https://github.com/PiSupply/PiJuice/commit/73d2aca98360c29eb38a57b623c1ee5a9df49ce0
Copy the current github version of pijuice_tray.py to /usr/bin/pijuice_tray.py.