PiSupply/PiJuice #391: Pijuice 12000mah mounting?

coltenxxx, 2019-10-10:
Not sure where else to ask this. But I had two questions about the Pijuice product. 

First of which, is there anyway I can implement a full shutdown of the pijuice and the raspberry pi 4B together without holding the 1 switch for 20 seconds? 

Second question, I ordered the 12000mah pijuice battery. Is there any way to mount this to the original pijuice hat? Whether aftermarket mounting or official? Wasn't sure if there was brackets, or a holder sold online.

tvoverbeek, 2019-10-10:
When only the PiJuice is powering the RPi (no external power connected) then you can tell the PiJuice to shut down the 5V to the Pi after xx seconds (typically 20 seconds) when you command the Pi shutdown yourself (either from the Desktop or by command `sudo poweroff`).
After the 5V is shut down the PiJuice goes to a low power mode. It never shuts down completely (unless the battery is empty).
When external power is connected the PiJuice remains on.

For the 12000mAH battery there are no mounting brackets available as far as I know. You connect the 3 wires to the screw terminals on the PiJuice. See https://learn.pi-supply.com/make/how-to-setup-connect-your-pijuice-battery/

coltenxxx, 2019-10-10:
Appreciate the feedback. Last question, how would I go about editing the parameters to alter the timer till shutdown on the pijuice, switch 1?

tvoverbeek, 2019-10-11:
Look at the Buttons tab in the PiJuice Hat Config on the GUI (or its equivalent in pijuice-cli):
https://github.com/PiSupply/PiJuice/tree/master/Software#pijuice-hat-config-buttons-menu
The parameter is the duration in milliseconds.

coltenxxx, 2019-10-12:
Thank you very much! That answers absolutely everything. I'll mark this as closed. Great help!

closed 2019-10-12 by coltenxxx