PiSupply/PiJuice #392: Can't Turn Off Flashing Blue/Green LED's Before Shutdown

norm-pi, 2019-10-11:
I've run the following script but it won't turn off the LED's. Any suggestions please?

#!/usr/bin/env python

import os
from pijuice import PiJuice

pijuice = PiJuice(1, 0x14)

# Remove Pi power after 30 seconds
pijuice.power.SetPowerOff(30)

# turn off LED's
pijuice.status.SetLedState('D1', [0, 0, 0])

# Shutdown
os.system("sudo halt")

tvoverbeek, 2019-10-11:
SetLedState only work for the function USER_LED not for CHARGE_STATUS (which is the default for LED 1)
Use SetLedConfiguration instead:
```
ledconfig = {'function': 'NOT_USED', 'parameter': {'r': 0, 'g': 0, 'r': 0}}
pijuice.config.SetLedConfiguration('D1', ledconfig)
pijuice.config.SetLedConfiguration('D2', ledconfig)
```
Note this is permanent. The LEDs will remain off when you boot again. You need to set the configuration again to turn them back on.

norm-pi, 2019-10-11:
Is there a document listing all the pijuice python functions?

tvoverbeek, 2019-10-11:
Ir is only partly documented at the end of the software README on github:
https://github.com/PiSupply/PiJuice/tree/master/Software#i2c-command-api
For now the best reference is the source file for the pijuice module:
https://github.com/PiSupply/PiJuice/blob/master/Software/Source/pijuice.py

norm-pi, 2019-10-11:
I got the following error??

$ python3 /home/osmc/myscripts/shutdown.py
Traceback (most recent call last):
  File "/home/osmc/myscripts/shutdown.py", line 13, in <module>
    pj.config.SetLedConfiguration('D1', ledconfig)
  File "/usr/lib/python3/dist-packages/pijuice.py", line 1343, in SetLedConfiguration
    d[3] = int(config['parameter']['b'])
KeyError: 'b'

norm-pi, 2019-10-11:
Found the error. 

In the first line of your code you have 'r' listed parameter: twice. The second 'r' should be a 'b' like the following;
ledconfig = {'function': 'NOT_USED', 'parameter': {'r': 0, 'g': 0, 'b': 0}}

Thanks for your help.

closed 2019-10-11 by norm-pi

tvoverbeek, 2019-10-11:
Good luck with your project.