PiSupply/PiJuice #393: User Function Not Executing

WadeVaresio, 2019-10-11:
I have defined a user script to be run on a no power event. System task is enabled. The goal of the script is to shut down the pi upon no power to prevent data corruption. I have deployed this script on a separate project and it runs as intended with the PijJice. 

With this new project, when a no power event occurs the pi will stay on from the PiJuice with the activity light rapidly blinking every few seconds, ultimately never properly shutting down. I am using the command line version, with firmware version 1.2. Power is supplied to the Pi and PiJuice via separate micro usb cables.

I believe this is possibly a permissions issue with the PijJuice install as when I set no power event to "SYS_FUNC_HALT_POW_OFF" it does not work either, resulting in the same behavior.

Am I missing something simple? Or have I installed PiJuice incorrectly?

```#!/usr/bin/python3

import os
from pijuice import PiJuice

pj = PiJuice(bus=1, address=0x14)

pj.status.SetLedBlink(led='D2', count=3, rgb1=[0, 0, 200], period1=1000, rgb2=[200, 0, 0], period2=1000)

# Remove power to PiJuice MCU IO pins
pj.power.SetSystemPowerSwitch(0)

# Remove 5V power to RPi after 60 seconds
# Wait for the timer to finish before restarting the board
pj.power.SetPowerOff(delay=60)

# Shut down the RPi
os.system('sudo shutdown now')```

tvoverbeek, 2019-10-11:
I am puzzled by ` Power is supplied to the Pi and PiJuice via separate micro usb cables`.
The no power event is only triggered when there is no power input on both the Pi and the PiJuice.
How are the two micro-usb cables connected to power?

WadeVaresio, 2019-10-11:
The two micro-usb cables are plugged in to a power strip, which when powered off cuts power to both the pi and PiJuice.

tvoverbeek, 2019-10-11:
The normal setup is to not connect power to both. Either power to the RPi (which will power the PiJuice and chsrge its battery via the 5V GPIO) or power to the PiJuice which will then power the Pi via the 5V GPIO.
Try with only one power connection.

WadeVaresio, 2019-10-11:
I will give the power to one a try, however this works fine for a different project of mine which leads me to believe that this is a software issue.

tvoverbeek, 2019-10-11:
I recommend to update your pijuice packages to the 1.5 version which also include the 1.3 firmware and then to update the firmware. There have been bug fixes and improvements to both the software and firmware.

WadeVaresio, 2019-10-11:
Tested on a PiJuice running firmware 1.3 and everything works as intended. Thanks for your help!

closed 2019-10-11 by WadeVaresio