PiSupply/PiJuice #394: SW3 Won't Activate Script

norm-pi, 2019-10-12:
Trying to get pijuice to work has been a struggle.

I have SW3 (press) set to run a script /home/osmc/myscripts/shutdown.py. If I run python3 /home/osmc/myscripts/shutdown.py the script runs OK. System Task is enabled, as per pijuice_config.JSON below.

{
  "user_functions": {
    "USER_FUNC1": "/home/osmc/myscripts/turnon.py",
    "USER_FUNC12": "",
    "USER_FUNC14": "",
    "USER_FUNC11": "",
    "USER_FUNC13": "",
    "USER_FUNC15": "",
    "USER_FUNC4": "",
    "USER_FUNC3": "",
    "USER_FUNC7": "",
    "USER_FUNC9": "",
    "USER_FUNC6": "",
    "USER_FUNC2": "/home/osmc/myscripts/shutdown.py",
    "USER_FUNC8": "",
    "USER_FUNC5": "",
    "USER_FUNC10": ""
  },
  "system_task": {
    "enabled": true
  }
}

norm-pi, 2019-10-12:
I've set SW3 to run the test logging script references in another post https://github.com/PiSupply/PiJuice/blob/master/Software/Test/user_func1.py and it logs OK when I press SW3. Just can't figure out why my script works from the interpreter but not from SW3.

ChristopherRush, 2019-10-12:
Did you make it executable?

norm-pi, 2019-10-12:
Yes I made it executable.

It took me all day but I finally figured it out. When my script is called from the interpreter like so;
"python3 myScript.py", it is specifying Python 3. When run from PiJuice when I press the button it uses the default version of Python which is Python 2.7 in OSMC. By changing the shebang to #!/usr/bin/python3 it works perfectly.

closed 2019-10-12 by norm-pi