PiSupply/PiJuice #396: Additional Data via Python

fictionalname, 2019-10-15:
Hello,

I have just started using a Pi Zero W with the PiJuice Zero and a 12000mAh battery and it all seems to be working fine so far.

I am using Python in a cron job to gather stats on the battery charge %, as well as other vitals from the Pi so I can graph them for monitoring my project.

I have noticed that within the PiJuice_CLI app, there is additional information like battery temperature and voltage in for the board.  

Is there any way of accessing these additional metrics via Python so I can add them to my stats?

Thanks for all your hard work and excellent product!

uGiFarukh, 2019-10-15:
After checking the GUI app, it seems that you need to call two different pijuice.status functions for getting the battery voltage and battery temperature. Make sure the PiJuice object is imported from the pijuice library. And, also the volt and temp variables are declared.

```
from pijuice import (PiJuice)

volt = pijuice.status.GetBatteryVoltage()
temp = pijuice.status.GetBatteryTemperature()
```

fictionalname, 2019-10-15:
Awesome!  Thanks for the info.

uGiFarukh, 2019-10-16:
Could you please confirm, if the provided code worked? If it worked, please close the issue or request the admins to close it. Thank You.

fictionalname, 2019-10-16:
I have just tested it, and yes this code does return the battery voltage (without the decimal point) and temperature.

Thanks.

closed 2019-10-16 by fictionalname

uGiFarukh, 2019-10-16:
Hi there,

Could you please use float type variables for both of the battery voltage and temperature outputs? Or probably use a type independent console print on the function itself to check the function return output first and then choose the variable type accordingly.