PiSupply/PiJuice #397: RPi keeps rebooting every 5 sec

cjdcordeiro, 2019-10-17:
Hi,

I'm having issues with the PiJuice Hat. 
On full charge (static blue LED), when I boot the RPi it just starts the bootstrap and goes straight into a loop.

I've tried plugging the RPi directly to power and monitor the battery with pijuice_cli, and everything seems normal. 

I've tried resetting everything to default and also upgrading firmware to v1.3.
Nothing changed. 

I'm running Raspbian Buster Lite on a RPi 3

tvoverbeek, 2019-10-17:
With power to the RPi and running pijuice_cli check on the 'System Events' page
what the state is of 'No power:'.  If it is enabled and set to e.g. SYS_FUNC_REBOOT
you get what you see: a reboot loop when no external power is applied.

cjdcordeiro, 2019-10-18:
@tvoverbeek no it's not enabled. All system events are disabled

tvoverbeek, 2019-10-18:
Any peripherals connected to the RPi? E.g a usb-disk spin-up at boot could draw too much power and causing a shutdown.
If not, it might be a defective unit. Take contact with your supplier  for a replacement.

cjdcordeiro, 2019-10-18:
nope. Just had a Pimoroni Blinkt. But removed it and the problem presisted

shawaj, 2020-12-05:
@cjdcordeiro did you manage to get this sorted now? Can we close this issue?

cjdcordeiro, 2020-12-07:
I didn't. but you can close it anyway