PiSupply/PiJuice #398: PiJuice shutting down Pi4 4GB inappropriately

Remotruker, 2019-10-19:
My PiJuice is shutting down the Pi4 after 7 minutes. This only happens when power is applied directly to the HAT.

Firmware: PiJuice-V1.3_2019_02_28.elf

My config from /var/lib:

  "system_task": {
    "enabled": true,
    "watchdog": {
      "enabled": false,
      "period": "10"
    },
    "wakeup_on_charge": {
      "enabled": true,
      "trigger_level": "1"
    },
    "min_charge": {
      "enabled": true,
      "threshold": "5"
    },
    "ext_halt_power_off": {
      "enabled": true,
      "period": "120"
    },
    "min_bat_voltage": {
      "threshold": ""
    }
  },
  "system_events": {
    "low_charge": {
      "enabled": true,
      "function": "SYS_FUNC_HALT_POW_OFF"
    },
    "low_battery_voltage": {
      "enabled": true,
      "function": "SYS_FUNC_HALT_POW_OFF"
    },
    "no_power": {
      "enabled": true,
      "function": "SYS_FUNC_HALT_POW_OFF"
    }
  },
  "user_functions": {
    "USER_FUNC1": "",
    "USER_FUNC2": "",
    "USER_FUNC3": "",
    "USER_FUNC4": "",
    "USER_FUNC5": "",
 "USER_FUNC6": "",
    "USER_FUNC7": "",
    "USER_FUNC8": "",
    "USER_FUNC9": "",
    "USER_FUNC10": "",
    "USER_FUNC11": "",
    "USER_FUNC12": "",
    "USER_FUNC13": "",
    "USER_FUNC14": "",
    "USER_FUNC15": ""
  }
}

ChristopherRush, 2019-11-14:
Are you stating here that the Pi 4 shuts down when you connect mains power to the PiJuice HAT after a period of 7 minutes or thereabouts?

ChristopherRush, 2019-11-14:
You have the `NO POWER` event checked, therefore as soon as you remove power to the Pi or PiJuice HAT it will shutdown.

Remotruker, 2019-11-14:
Christopher, correct. The pi shuts down while mains are still connected. 

For the No Power event, correct, and understood. 

This pi will be going into a vehicle so the goal is power up when mains applied, and power down when mains removed.

ChristopherRush, 2019-11-14:
I have just setup as per your config file above and the Raspberry Pi stayed on for 1hour+ without any sudden shutdown.

I would suggest to reset to factory settings and then use `No Power` to do a safe shutdown when power is lost then `Wakeup On Charge` to boot when power has been restored. Nothing else needs to be set to achieve what you want.

philicibine, 2020-01-22:
I am experiencing similar behaviour with my pi4 (setup in my van).  Although it isnt 7 minutes.. my pi is also not shutting down safely, rebooting regularly and randomly and sometimes the pijuice isnt registering on i2c.  

I have it powered by a 5amp 12v to 5v usb stepdown module.  So the power should be ample.. but i'm still having these issues.  

I thought it maybe due to the voltage fluctuations caused by starting the van whilst the pi is booting, but I have tried a few different wires and ways of powering the pi and it always seems to do the same thing.  

How can i reliably power this in my vehicle??

Remotruker, 2020-01-22:
I have found that I need to power thru the hat, and only use the auto-shutdown routine thru loss of power. Do not use terminal or gui. 

My 7 min shutdowns have stopped after doing a fresh noobs and starting over with the above routine. 

Have A Day- Make It A Great One

shawaj, 2020-10-24:
@Remotruker @philicibine do you need any further assistance?

shawaj, 2020-12-05:
@Remotruker @philicibine closing this issue as no reply in over a month.

Reply if you don't agree

closed 2020-12-05 by shawaj