PiSupply/PiJuice #399: Turn off Pi when battery is at 10%

markb9919, 2019-10-19:
Hi 

I am using raspberry pi and pi juice with solar panel. I want to be able to stop the pi coming on when the battery level is around 10%. 

I need the pi to come back on when the level is above 10%.

I am using a schedule to turn the pi on every hour.

Any suggestions would be great.

Thanks

tvoverbeek, 2019-10-20:
On the 'System Task' panel enable 'Minimum charge' and set the threshold to 10.
On the 'System Events panel enable 'Low charge' and link it to a user function or directly to SYS_HALT_POW_OFF if that suits your app better.
In the user function case you could set a flag which is picked up in your hourly app and then at shutdown suspends the hourly schedule.

On the 'System Task' panel enable 'Wakeup on charge' and set the threshold above 10, e.g. 25.
Then when the battery lcharge level reaches 25% the Pi wakes up. You should check the charge level at boot and when above 25 and the hourly schedule is disabled, enable the schedule again.

ChristopherRush, 2019-10-23:
@markb9919 Did this work for you?

ChristopherRush, 2019-11-10:
@markb9919 Closing this issue as no response since my last comment. If you still need help then please reopen.

closed 2019-11-10 by ChristopherRush