PiSupply/PiJuice #400: Service does not start on boot

gmnoybus, 2019-10-22:
What is needed to get the service to start automatically whenever the pi is booted? 

Here is all the configuration and information I have gathered in trying to solve this problem.

HW: Pi3B+
Software: Raspian Buster
headless configuration, all access via ssh, user pi
install via sudo apt-get install pijuice-base
I believe, but cannot guarantee, I followed all the install directions on this site.

battery is not yet fully charged

service did not start after installation.
service does not start on reboot.
/dev/i2c-1 exists
i2ctools does not exist

pi@piholehost:/dev $ systemctl status pijuice.service
pijuice.service - PiJuice status service
   Loaded: loaded (/lib/systemd/system/pijuice.service; disabled; vendor preset:
   Active: inactive (dead)

service will start using systemctl start pijuice.service

gmnoybus, 2019-10-22:
Thank you.
Standard Raspbian Lite. No modifications.
Yes, pihole is installed and quite functional.
In what way does that affect the success of a pijuice install? And, what may happen to pijuice if I choose to install other software?

Seems things are now working. Perhaps an addition to your install instructions is in order if this is a common problem (as indicated by your quick diagnosis)?

pi@piholehost:~ $ sudo systemctl enable pijuice.service
Created symlink /etc/systemd/system/multi-user.target.wants/pijuice.service → /lib/systemd/system/pijuice.service.
pi@piholehost:~ $ sudo systemctl start pijuice.service
pi@piholehost:~ $ systemctl status pijuice.service
● pijuice.service - PiJuice status service
   Loaded: loaded (/lib/systemd/system/pijuice.service; enabled; vendor preset: 
   Active: active (running) since Tue 2019-10-22 10:47:03 PDT; 1h 1min ago
 Main PID: 992 (pijuice_sys.py)
    Tasks: 1 (limit: 2200)
   Memory: 5.0M
   CGroup: /system.slice/pijuice.service
           └─992 /usr/bin/python3 /usr/bin/pijuice_sys.py

Oct 22 10:47:03 piholehost systemd[1]: Started PiJuice status service.
pi@piholehost:~ $ sudo shutdown now -r
pi@piholehost:~ $ Connection to 192.168.7.89 closed by remote host.
Connection to 192.168.7.89 closed.

//Reconnect via ssh

pi@piholehost:~ $ systemctl status pijuice.service
● pijuice.service - PiJuice status service
   Loaded: loaded (/lib/systemd/system/pijuice.service; enabled; vendor preset: 
   Active: active (running) since Tue 2019-10-22 11:49:27 PDT; 2min 29s ago
 Main PID: 490 (pijuice_sys.py)
    Tasks: 1 (limit: 2200)
   Memory: 9.8M
   CGroup: /system.slice/pijuice.service
           └─490 /usr/bin/python3 /usr/bin/pijuice_sys.py

Oct 22 11:49:27 piholehost systemd[1]: Started PiJuice status service.
pi@piholehost:~ $

tvoverbeek, 2019-10-22:
The post install script which is run when you install the pijuice-base package does enable the service and starts it. This does not seem to have happened in your case, which is why I thought it might be a non-standard installation. Also the pijuice-base package has a dependency on i2c-tools. Hence the i2c-tools package should have been installed when you installed pijuice-base.
Bit difficult from here to say why this did not work for you.

gmnoybus, 2019-10-22:
Works now so we can move on. Thanks.

closed 2019-10-22 by gmnoybus