PiSupply/PiJuice #401: DATA_CORRUPTED errors

BigEpidemic, 2019-10-23:
I've run i2cdetect -y 1 and see 14 and 68. Battery on the HAT menu shows DATA_CORRUPTED most of the time, but if I bring up Configure Hat and set i2c address 1 to 14 and address 2 to 68 and select either 50 or 52 in the ID_EEPROM_ADDRESS it immediatly displays a DATA_CORRUPT dialog but when I dismiss it  the combo value switches back to DATA_CORRUPT but the battery displays the current value for a few seconds, then switched back to DATA_CORRUPT. 

Firmware shows 15.15 installed 

I've tried reset to default conditions a few times with no success.

uGiFarukh, 2019-10-23:
Firmware 15.15 that is so weird. The current version of firmware is 1.3 . Can you please provide some screenshots of the situation?

BigEpidemic, 2019-10-23:
Problem solved. I found the comment that I needed to change the config.txt
for RP4 and made that change. Now it works great. That really should be
easier to find...

tvoverbeek, 2019-10-23:
Yes, it is a known problem on the Pi4. The firmware developer is working on a fix, so in the future the Pi4 should not need to fall back to the old i2c driver (i2c_bcm2708).
Anyway, glad you solved it yourself.
Closing this issue, Reopen it if you do not agree.

closed 2019-10-23 by tvoverbeek

marza91, 2020-04-16:
@BigEpidemic @tvoverbeek 
I think I'm having this exact issue, fresh Raspberry Pi 4 with newly flashed Raspbian. Could you provide a link the where it says what needed to be changed in the config.txt file?

marza91, 2020-04-16:
I'm guessing it is this one(?) https://github.com/PiSupply/PiJuice/issues/327
I just packed everything up, so I'll need to test this tomorrow, but hopefully it'll work

tvoverbeek, 2020-04-17:
@marza91 
WIth a fresh Raspbian and the 1.6 packages installed on a Pi4 you get DATA_CORRUPTED errors when using the Pijuice Firmware as delivered (firmware version < 1.4).
- Add the line `dtoverlay=i2c-bcm2708` to /boot/config.txt and reboot. This should get rid of the DATA_CORRUPTED errors.
- Check the firmware version in the GUI/CLI. With the 1.6 pijuice-base package installed it should prompt you to install the 1.4 firmware.
- After upgrading the firmware to 1.4 you can delete the line `dtoverlay=i2c-bcm2708` from /boot/config.txt again.

This only applies to the Pi4, since it has different I2C hardware than the previous Pis.

marza91, 2020-04-17:
This is what I hoped would fix it as well, but after trying today it seems adding the dtoverlay line to my boot config file does not make the DATA_CORRUPTED errors go away. There was already a line with dtoverlay=vc4-fkms-v3d in my config, so I also tried with that line commented out, but that didn't help either.

Also, just to mention everything, I have an official Pi 7" touch display connected to the top of the PiJuice GPIO pins (5v, ground, GPIO2 and GPIO3). Could that me causing issues? I can try to remove it, just have to do some more setup for another screen and/or ssh..

(Should i open a new issue rather than commenting on this one?)

marza91, 2020-04-17:
Also, I couldn't find the instructions for doing a firmware upgrade from console anywhere. Probably me being blind, but I will gladly try that as well since it's not possible in the GUI.

marza91, 2020-04-17:
Okay, I managed to get it working, and firmware updated!
I removed the two cables connected to GPIO2 and 3, and everything worked out! I thought those were used for the touch controls on the screen (I've had it that way since I got the screen a few years back), but the touch controls are still working without them... Not sure where I read to put them there either, cause the docs doesn't mention them as far as I can see... But that's not related to this issue, so I'll look elsewhere for that.

tvoverbeek, 2020-04-17:
The connections to GPIO2 and 3 (the I2C bus) are only needed for the original model B and A.
See https://www.raspberrypi.org/documentation/hardware/display/README.md and especially https://www.raspberrypi.org/documentation/hardware/display/legacy.md
These connections messed up the I2C communication with the PiJuice.