PiSupply/PiJuice #402: Stop blinking LEDs on system shutdown

mminehanNZ, 2019-10-24:
Just got my PiJuice a few days ago.  Very pleased compared to another product I was using which was buggy and poorly supported.
I am powering the PiJuice using the offical Raspberry Pi PoE HAT.  Seems to work well.
I use the LED (D2) as a status indicator of running services (InfluxDB and Grafana).  Different colour blinking patterns correlate to which service are and aren't running, e.g. green for influx, blue for grafana, and red if one or the other is not running.  The LEDs are set using a crontab python script that run every 5 minutes.
The question I have is how to stop the blinking on system shutdown?  It seems the LEDs continue to blink on reboot even though the script has not been executed again.
I tried putting a bash script, that calls another python script turn off the LED, into usr/lib/systemd/system-shutdown but the script never runs.  I think the python modules shutdown before the script is executed.
Anyone have any suggestions on how to turn off the LEDs on shutdown or restart?
Thanks, Marty.

tvoverbeek, 2019-10-24:
Use your python script and make it executable (`chmod +x  <your-script>`).
Make sure the first line of your script is `#!/usr/bin/python3`.
To execute this script at shutdown/restart make it a service for systemd.
How to do this is described here: https://askubuntu.com/questions/416299/execute-command-before-shutdown-reboot
Look at the section "How to Do it with Systemd (it's easier)"
Although it is for Ubuntu it will work the same way for Raspbian.

mminehanNZ, 2019-10-26:
Hi @tvoverbeek ,
Thanks for your advice.  I got it working using systemd technique that you linked to.  Ended up dropping the shell script and just ran the python script directly.  This my service:

> [Unit]
> Description=Run Scripts at Start and Stop
> DefaultDependencies=no
> Before=shutdown.target reboot.target halt.target
> 
> [Service]
> Type=oneshot
> ExecStart=/usr/bin/env python /home/pi/bin/ledoff.py
> 
> [Install]
> WantedBy=halt.target reboot.target shutdown.target

Thanks again for your help.

closed 2019-10-26 by mminehanNZ