PiSupply/PiJuice #403: Need RP4 to safe shutdown when power lost (with delay) and auto-start when power applied

BigEpidemic, 2019-10-25:
I can't get the user_func1.py solution you propose to even fire. No log is ever generated in the folder. I've set permissions on the file to execute. When I pull power from either the USBC or the micro-usb the script never seems to execute. I'm pretty new to Linux so maybe I'm doing something wrong.

Is what I'm trying to do possible?

ChristopherRush, 2019-10-25:
Just tested this and the log file was created when I removed power from the Raspberry Pi. What OS are you running? I did `sudo chmod +x` on the script before hand.

tvoverbeek, 2019-10-26:
As @ChristopherRush wrote the script needs to be executable.
Moreover the owner of the script (normally pi) has to be a member of the pijuice group.
The default user (pi) is already a member of the pijuice group.
Assuming you are using the GUI:
On the 'System Events' tab you have to enable 'No power:' and link it to USER_FUNC1.
On the 'User Scripts' tab you have to specify the path for 'USER FUNC1:' e.g. '/home/pi/user_func1.py'.
Most likely you already did this, but did not make the script executable.
The sample script will run under both python2 and python3.
If you later add pijuice commands to the script you have to use python3.
Suggest to change the first line of the script to `#!/usr/bin/python3`

ChristopherRush, 2019-11-10:
@BigEpidemic Do you still need help with this?

BigEpidemic, 2019-11-12:
No. Thank you for your attention to this. I've switched to another product.

closed 2019-11-14 by ChristopherRush