PiSupply/PiJuice #404: pijuice_cli crashing on buster lite

zubenubi, 2019-10-26:
Hi,

New pi4 with buster lite.

When I run pijuice_cli, the menu loads, but if I try to go into a sub-menu, it crashes with the following output:
`Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 2007, in <module>
    loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 286, in run
    self._run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 384, in _run
    self.event_loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 788, in run
    self._loop()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 825, in _loop
    self._watch_files[fd]()
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 404, in <lambda>
    event_loop, callback, self.get_available_raw_input())
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 502, in parse_input
    callback(processed, processed_codes)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 411, in _update
    self.process_input(keys)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 511, in process_input
    k = self._topmost_widget.keypress(self.screen_size, k)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 595, in keypress
    *self.calculate_padding_filler(size, True)), key)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 1590, in keypress
    key = self.focus.keypress(tsize, key)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 2271, in keypress
    key = w.keypress((mc,) + size[1:], key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/listbox.py", line 999, in keypress
    key = focus_widget.keypress((maxcol,),key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/wimp.py", line 540, in keypress
    self._emit('click')
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 460, in _emit
    signals.emit_signal(self, name, self, *args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 265, in emit
    result |= self._call_callback(callback, user_arg, user_args, args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 295, in _call_callback
    return bool(callback(*args_to_pass))
  File "/usr/bin/pijuice_cli.py", line 1907, in item_chosen
    callback()
  File "/usr/bin/pijuice_cli.py", line 191, in __init__
    self.main()
  File "/usr/bin/pijuice_cli.py", line 251, in main
    status_args = self.get_status()
  File "/usr/bin/pijuice_cli.py", line 206, in get_status
    general_info += ", %3d°C, %s" % (temp, status['battery'])
KeyError: 'battery'
`

tvoverbeek, 2019-10-26:
On the Pi4 you need to use the old i2c driver i2c-bcm2708.
With the default i2c-bcm2835 driver the i2c communication to the Piuice does not work on the Pi4.
We are working on a fix, but for now add the following line to the `[pi4]` section of /boot/config.txt:
`dtoverlay=i2c-bcm2708`
It it then still does not work, let us know.

zubenubi, 2019-10-26:
Hi, yes I found a post describing the change that needed to be made to /boot/config.txt and it seems to be working well now.

Thanks for confirming this is the fix I needed

tvoverbeek, 2019-10-26:
OK. Glad it works. Closing this issue

closed 2019-10-26 by tvoverbeek