PiSupply/PiJuice #405: pijuice in kali

carlyx1987, 2019-10-26:
I can't install pijuice gui in kali linux.. can you help me step by step what to do.. thanks..

shawaj, 2019-10-26:
Have you tried anything at all so far?

carlyx1987, 2019-10-26:
I tried installing it manually and got my os wreck.. Im new to linux huhuhu

carlyx1987, 2019-10-26:
I tried the sudo apt-get install pijuice-gui and it says E: package not found sumthing

carlyx1987, 2019-10-27:
E: unable to locate package

ChristopherRush, 2019-10-27:
I have been successful in installing PiJuice Base with Kali, I will post install process shortly.

ChristopherRush, 2019-10-27:
Update `sudo apt update` then install dependancies `sudo apt install i2c-tools` `sudo apt install lua5.3` Add user `root` to group `pijuice` run `sudo visudo` add `root ALL=(pijuice) ALL` save and exit. Clone the repository `git clone https://github.com/PiSupply/PiJuice.git` install pijuice base from `cd PiJuice/Software/Install` run `dpkg -i pijuice-base_1.5_all.deb` Fix broken packages with `sudo apt --fix-broken install` reboot and you should be able to run `pijuice_cli` from the terminal. 

Once pijuice-base has been installed you can then install the GUI by running `dpkg -i pijuice-gui_1.5_all.deb` GUI requires `Python3-tk` install with `sudo apt --fix-broken install` reboot. PiJuice will be added to the application menu and you can run from there or from terminal with `pijuice_gui`.

ChristopherRush, 2019-10-27:
[image]
[image]
[image]

carlyx1987, 2019-10-27:
I'm running kali on a raspberry pi 4. thanks for the help my good sir. ill try this fix.. now and update you..

tvoverbeek, 2019-10-27:
@carlyx1987 One additional remark since you are using a pi4. If Kali uses the same kernel module for i2c (i2c-bcm2835) as Rapbian i2c communication to the pijuice fails. Symptoms: After starting pijuice_cli select Status and pijuice_cli crashes. In the GUI you will see 'COMMUNICATION ERROR' instead of actual values.
In /boot/config.txt (or Kali's equivalent) add the line `dtoverlay=i2c-bcm2708` to force the system to use the previous i2c driver (but maybe Kali uses i2c-bcm2708 by default).
@ChristopherRush on what Pi did you run Kali? Pi 3b+ and before do not have this problem.

carlyx1987, 2019-10-27:
Ok sir @tvoverbeek .. Sir @ChristopherRush can you help me with this part? add user root to group pijuice run sudo visudo add root ALL=(pijuice) ALL save and exit.

ChristopherRush, 2019-10-27:
@tvoverbeek Ran the above on Pi 4 and yes I had to add `dtoverlay=i2c-bcm2708` for some reason I had to do this as a last step rather than before the install. @carlyx1987 From the terminal window run `sudo visudo` then type in somewhere `root   ALL=(pijuice) ALL` CTRL+O to save CTRL+X to exit.

carlyx1987, 2019-10-27:
I can just do CTRL+O but I dont have the option to exit.

ChristopherRush, 2019-10-27:
Sorry CTRL+O then Enter to save CTRL+X then enter to quit

carlyx1987, 2019-10-27:
how to enable i2c? When I open pijuice_cli it says no pijuice found or Failed to use i2c bus. Check if i2c is enable

ChristopherRush, 2019-10-27:
Edit /boot/config.txt `sudo nano /boot/config.txt` then add the line `dtparam=i2c_arm=on`

carlyx1987, 2019-10-27:
It is still not working for me.. when I open pijuice_cli still says no pijuice found and when I click pijuice settings in application settings nothing happens.. huhuhuh

shawaj, 2019-10-27:
@ChristopherRush just a thought based on this issue. Do you think it would be a good idea to have an area in the github where we save this kind of info - install instructions and scripts for other OSs than raspbian?

ChristopherRush, 2019-10-27:
@shawaj 100% yes @carlyx1987 Try `sudo usermod -aG i2c root` then reboot again

carlyx1987, 2019-10-27:
OMG!!! Thank you guys its working now.. Thanks for the time and effort. God bless you guys!

ChristopherRush, 2019-10-27:
Super, glad you got it working!

tvoverbeek, 2019-10-27:
@carlyx1987 It is working, but if you want to use user functions these will fail.
The pijuice service which starts these functions does not allow scripts owned by root, even if root belongs to the pijuice group.
Kali by default only has the root user.
More details can be provided if you need them.

carlyx1987, 2019-10-27:
@tvoverbeek but can I atleast change settings like th long press etc? only the scripts will fail?

tvoverbeek, 2019-10-27:
@carlyx1987 Yes, changing all the other settings will work.

carlyx1987, 2019-10-27:
so sad. :(

ChristopherRush, 2019-10-27:
@tvoverbeek Is there a workaround for this?

carlyx1987, 2019-10-27:
Is there a function that is I shutdown RPI the pijuice will turn off also? planning to built a PI laptop with pelican case..

tvoverbeek, 2019-10-27:
@ChristopherRush Yes, create a normal unprivileged user and put the scripts there owned by the unprivileged user. Of course also this user has to be added to the pijuice group.

carlyx1987, 2019-10-27:
Can you give a tutorial for a beginner like me sir @tvoverbeek ?

tvoverbeek, 2019-10-27:
I do not have Kali, maybe @ChristopherRush can test it on his Kali and give you instructions.
Look at the adduser command.

ChristopherRush, 2019-10-27:
@carlyx1987 Under "System Task" enable "Software Halt Power Off" to turn off the PiJuice when shutting down the Raspberry Pi.

carlyx1987, 2019-10-27:
@ChristopherRush Thanks sir..

carlyx1987, 2019-10-27:
@ChristopherRush  I tried to install it in 64bit kali its not working..

carlyx1987, 2019-10-27:
@ChristopherRush  it says Failed to execute command "/usr/bin/pijuice_gui". . any idea how to fix this?

ChristopherRush, 2019-10-27:
I will take a look tomorrow.

ChristopherRush, 2019-11-10:
@carlyx1987 I was able to execute the pijuice_cli.py but not pijuice_cli in Kali 64. Not sure why, could be a build issue since its 64-bit.

tvoverbeek, 2019-11-11:
The pijuice_cli and pijuice_gui wrapper programs are 32-bit. If Kali 64 does not support executing 32-bit programs you need to recompile them for 64-bit. See the source and build script in https://github.com/PiSupply/PiJuice/tree/master/Software/Source/Setuid-Wrapper

sapster77, 2019-11-28:
Hi,
Similar issue in Kali. I get a "bash: /usr/bin/pijuice_cli: no such file or directory" when running PiJuice_cli or gui. I have been through all of above and am running it from a low privileged user "pi", which has been added to PiJuice and i2c group. The install went through without errors and "systemctl status PiJuice.service" returns "active running" (green dot). 
I'd be very thankful for any ideas?

sapster77, 2019-11-28:
This on a RPi 3b+ by the way..

sapster77, 2019-11-28:
We'll I can run the scripts directly "pyhton3 pijuice_cli.py" and it works just fine, so it's probably a user rights issue.

ChristopherRush, 2019-11-28:
Is this 64-bit Kali?

sapster77, 2019-11-28:
yep - 64.

ChristopherRush, 2019-11-28:
In that case read Ton's comments above.

sapster77, 2019-11-28:
Thanks I'll give that a go. Sorry about that - don't know how I could miss that comment...

ChristopherRush, 2020-02-24:
Documented install process - https://github.com/PiSupply/PiJuice/commit/7d629ab3957b2bda80d6cb1c7eb67ff23b88b05b

closed 2020-02-24 by ChristopherRush