PiSupply/PiJuice #406: Power on RaspberryPi with GPIO push button

larryval, 2019-10-27:
Hi!

My RaspberryPi and the PiJuice will be in a box, and I will not have access to the PiJuice button. I work with Raspbian Buster Lite (no GUI) and PiJuice-base v1.3.

After reading all the software documentation (and especially the python API), I can not find a solution to what I want to do:
When the Pi raspeberry is power-off, is it possible to start it with a push button connected to a GPIO pin? exactly what does SW1 by default?
thank's !

tvoverbeek, 2019-10-27:
Note that you can connect a push button to J5.
See https://github.com/PiSupply/PiJuice/tree/master/Hardware#buttons
The push button then has the same function as SW1

larryval, 2019-10-27:
holly s***t ! ! !  , that was in front off my eyes, I'll try this right now, thanks a lot @tvoverbeek  !

closed 2019-10-27 by larryval