PiSupply/PiJuice #407: Temperature issues with PiJuice, BP7X_1820 and the PiJuice tall case

tomn46037, 2019-10-28:
I have been trying to get the PiJuice working with a Raspberry Pi 3b+. I've got the PiJuice, a BP7X_1820 and a PiJuice case.  While it is working, I think these temperatures are too high.

The problem I'm seeing is that at idle in the PiJuice case, the battery is about 53 degrees C. Sounds warm to me. If I run a stress test, that battery temperature climbs pretty quickly to 60 or more. (I stopped at 60 as I saw something about a recommended maximum of 60 degrees C.)

Am I doing something wrong? Should I be seeing temperatures this high?

ChristopherRush, 2019-11-07:
What is the temperature like when the Pi and PiJuice is outside the case? Do you see normal temperature values?

ChristopherRush, 2019-11-14:
@tomn46037 Any update to this?

closed 2019-11-25 by ChristopherRush